iNavFlight/inav #4478: switching from Alt hold to HOLD ,Quad crash

sincoder, 2019-03-09:
## Current Behavior

[](https://youtu.be/vHIg_sz80KM)
## Steps to Reproduce
Take off at Angle + ALT Hold , switch to HOLD, Lost control and quad crash

## Expected behavior

## Suggested solution(s)
seems like a Race Condition Bug?

## Additional context
did a nice HOLD yesterday,  But crashed today
---
# dump

# version
# INAV/OMNIBUSF4V3 2.1.0 Feb 25 2019 / 16:53:57 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
[... 941 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 460
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

stronnag, 2019-03-09:
Let me see ... 
```
set nav_extra_arming_safety = OFF
```
And if you 

* `set nav_extra_arming_safety = ON`
* Provide a blackbox log

We can see what's really the case.

sincoder, 2019-03-09:
@stronnag always set nav_extra_arming_safety = OFF and task off at HDOP 2.0 , seems OK , No BB log ,need some days to repair the quad and fly again

teckel12, 2019-03-09:
@sincoder HDOP of 2.0 may not be as good as you think. It depends on if your GPS is reporting HDOP or PDOP. Also, I've seen low HDOP but the altitude is a hundred feet off or it's showing the quad is moving 10mph while unarmed.

Basically, if your using only HDOP, you're probably not as safe to fly as you think. I'd **HIGHLY** suggest `nav_extra_arming_safety = ON` and also verify GPS altitude is correct before arming.

sincoder, 2019-03-10:
confirmed, hardware issue, ,one motor wire broken

closed 2019-03-10 by sincoder