iNavFlight/inav #4481: LED Strip Aux Control

MartinHugh, 2019-03-09:
Is it possible to control nominated LEDS on the LED_STRIP interface from the value in an Aux channel.

For background I was interested in indicating whether or not Auto Launch was active using flashing LEDs, as a safety measure.

thanks

teckel12, 2019-03-10:
@MartinHugh Other modes can be used to provide indication, can't see why Auto Launch can't also be an indicator.  I don't fly fixed wing so I wouldn't be the best to write/test this.  But it seems fairly straight-forward, and possibly other flight modes could also be incorporated at the same time.

MartinHugh, 2019-03-11:
thanks for your comments @teckel12 .

Martijn0001, 2021-07-05:
Hi there, this seems to be an older track. Is this request still active? I would also appreciate a function where we can control leds based on channel value. Maybe even the ability to create LED profiles and allow selecting one by a certain channel.

closed 2023-03-06 by DzikuVx