iNavFlight/inav #4486: RTH procedure Landing mode issue

Jetrell, 2019-03-09:
Have any of the Dev's ever tested the LAND function, that is supposed to be part of the failsafe RTH procedure?
[image]

I'm not referring to the failsafe LAND procedure that uses a set throttle and timer. It does work if selected.

I'm referring to the way that the quad should land in RTH procedure by barometer and IMU, if the GPS signal is lost.

**(From the failsafe wiki)**
 **_If any required navigation sensor becomes unavailable during a Return to Home (eg. loss of GPS fix), an emergency landing similar to SET-THR but with barometer support will be performed after a short timeout. An emergency landing would also be performed right when the failsafe is triggered if any required sensor is reported as unavailable._**

I have tested this  function many times. And all it does is drop the throttle down to zero, like DROP procedure.  It does not revert to the LAND procedure Throttle value, nor does it work with the Timer under that procedure.

I don't understand why it doesn't land the quad by the barometer and IMU, the same way that a RTH auto land would occur. Then shut the motors off, like it does under the RTH procedure with GPS lock.
I realize that it would not be able to hold position without GPS. But at least it would still land under power and shut the motors off after touch down.
Instead it crashes to the ground. Then will runs your battery flat if it has no RF link, because the motors keep spinning if AIR mode was enabled with RTH.

This bug has been here for many releases and needs to be addressed for safety reasons.

If the you guys are concerned about how to test these issues. I have made a servo operated jig that covers the GPS in flight. To simulate the issue. 
I don't mind doing the testing if you provide code that has some likely hood of a fix.

digitalentity, 2019-03-10:
Do you have blackbox log to confirm the described behavior? 
Last time I tested emergency landing (landing in RTH when GPS is lost) it was working as intended, so this has to be either a recently introduced bug or an edge case

Jetrell, 2019-03-10:
No. The FC in my test quad has a bent chassis on its card holder. I use this quad for tests in hope that it will confirm any setting or bug issue, so to to avoid damage occurring to my larger 10" machines if the same thing was to happen.

A question I also have surrounding this particular function, that the manual doesn't make completely clear.
If RTH procedure is selected and a GPS was not used...Or the quad was flown before a lock was establish and a failsafe condition did occur. Would iNav instigate a landing, based on the RTH procedure using the  IMU and Barometer?
Or would it revert to the LAND or DROP procedure in these cases?

Thanks for the reply

Jetrell, 2019-03-16:
> If RTH procedure is selected and a GPS was not used...Or the quad was flown before a lock was establish and a failsafe condition did occur. Would iNav instigate a landing, based on the RTH procedure using the IMU and Barometer?
> Or would it revert to the LAND or DROP procedure in these cases?

@digitalentity  Do you know the answer to the question from a few days back? I figured that if anyone did. It would be you :-)

At this stage. I'm not willing to take a risk with any of my larger quads that have a data logger on their FC's. After seeing how it worked out with my test rig quad.

If this is a bug, I can understand why it would slip by unnoticed in 99.9% of cases. Because the odds of a GPS failure under a failsafe condition is very slim. But that's when you need a backup system to work the most!

But I did run a ground based test with one of the others quads.

I am not sure how conclusive the results would be though. Due to the FC not seeing any barometeric changes that may effect the way the throttle would react when failsafe is triggered without any GPS lock?
So its only speculation without your insight.

The test I performed was by walking around with the quad once it had a GPS lock and having the motors running around 45% throttle.
Then I turned the radio transmitter off to forced a failsafe. Under this condition the quad would instantly increase the motors throttle to full and start to climb to RTH fly home altitude. Seems normal.

I then tried it two other ways. Once by triggering failsafe without a GPS lock. And the other by covering the GPS and taking away it lock while in failsafe.
The first caused the throttle to drop to zero and Air mode kept the motors spinning. Then once I placed the quad on the ground there was still no timeout or motor disable. 
And the second just kept trying to climb out to RTH fly home altitude. Not sure what would have happened if I had let it get there, with no GPS signal :-/ !

As I said. These are not totally conclusive. And even less so, without me knowing how the code functions under these conditions.
Can you inform me of what you think?

digitalentity, 2019-03-16:
TL;DR: Still impossible to debug without BB logs and config dump.

I've looked at the code and it should execute emergency landing procedure if GPS is not available during RTH (regardless of failsafe or not). You may want to test disabling GPS lock while simulating normal RTH (not activated by failsafe).

There is a caveat - GPS status is ignored in certain cases when doing initial RTH climb - this is controlled by `nav_rth_climb_ignore_emerg` setting. If you have it enabled it explains the continued attempts to climb.

Please bear in mind that it's next to impossible to troubleshoot without any actual data. Please retry your tests with BB logging enabled and provide a config dump.

Jetrell, 2019-03-16:
Thanks @digitalentity 

I went out and simulated the tests with a quad that has a logger. In the same way I described earlier.
With regard to `nav_rth_climb_ignore_emerg ` I never use it on quads. It just leaves farther to fall if something like this happens ;-).
The logs are labeled with the tests I performed. 

Test 1 performed as it should. Its just here as a reference.

The power on the TX and FC was cycled between all tests. And the quad was lower to the ground and left sit for longer than 20sec, to see if the FC would disable the motors; on tests 2 and 3.
FS was enable with both TX off and by mode switch in each test.

There was quite a variation in the results. Comparing FS TX off and FS mode switch selection. Note in test 2. The motors shut down immediately after FS, but Air mode stays active in RTH. Never disabling. 

As a first side note. I included a text doc on the way Crossfire FS reacts differently depending on whether it is set to CUT of FAILSAFE - POS.  I presently have it set to FS -POS. Because of safety reasons that can be seen in the Doc.  I thought this might be effecting this topic.

And another side note. These logs will also show up the Matek M8Q delayed acknowledgement time, on power up. If it helps with the other post. 

[3) With GPS fix, FS act both TX off and mode, GPS lock taken away.TXT](https://github.com/iNavFlight/inav/files/2974602/3.With.GPS.fix.FS.act.both.TX.off.and.mode.GPS.lock.taken.away.TXT)
[Crossfire FS quirk.txt](https://github.com/iNavFlight/inav/files/2974603/Crossfire.FS.quirk.txt)
[Diff setup.txt](https://github.com/iNavFlight/inav/files/2974604/Diff.setup.txt)
[1) With GPS fix and failsafe both TX off and mode.TXT](https://github.com/iNavFlight/inav/files/2974605/1.With.GPS.fix.and.failsafe.both.TX.off.and.mode.TXT)
[2) No GPS lock, FS both TX off and mode.TXT](https://github.com/iNavFlight/inav/files/2974606/2.No.GPS.lock.FS.both.TX.off.and.mode.TXT)

digitalentity, 2019-03-17:
Thanks for the data. I'll have a look in the next few days.

Jetrell, 2019-03-22:
Did you get the time to look at the logs @digitalentity ?

Jetrell, 2019-03-27:
Failsafe Landing mode has another bug that I found.  
[image]

It is related to the way `failsafe_off_delay ` interacts with **RTH Procedure** failsafe navigation. When Stick Arm/Disarm is used.

e.g. If the `failsafe_off_delay` is set to 20 seconds. 
As soon as failsafe triggers the RTH procedure. The `failsafe_off_delay` timer starts counting down. 
Once it has expired, the RTH navigation is aborted by the FC and the motors are shut off. And it once more drop to the ground after it just finished climbing to RTH fly home altitude.  That can be a LONG fall ! Guaranteed to trash most multirotors.
I have found nothing in the Wiki that warns of this.

I find this ironic. That it will disable the motors under both the above mentioned conditions. BUT if a crash occurs when a FixedWing or Multirotor is in an Auto Navigation mode (RTH or Waypoint) **with no RF link**. Then it DOES NOT disable the motors. Instead it either flattens the battery; burns out your ESC/Motors or could potentially set a whole field on fire. 

I figure that most people, are lucky that they have never had a GPS failure in Failsafe. Or if they have...They never got to the bottom of what caused their crash.

@digitalentity This three things should be CORE redundancy features. Flight controller software that potentially trashes our model, if a hardware failure occurs, isn't much good :-1: 

I know this is an open source repository. But surely things like this, should be prioritized over the long lists of new features on every release?

Jetrell, 2019-04-01:
Any word @digitalentity ?
Were you able to confirm these tests?

digitalentity, 2019-04-06:
>It is related to the way failsafe_off_delay interacts with RTH Procedure failsafe navigation. When Stick Arm/Disarm is used.

`failsafe_off_delay` is ignored when failsafe is executing RTH. If you have proof of the otherwise - please share your data.

>@digitalentity With all respect. This three things should be CORE redundancy features. Flight controller software that doesn't assist our model, if a GPS hardware failure occurs, defeats the purpose of having a FC.

You seem to misunderstand the purpose of failsafe - to make an attempt to save the aircraft, not guarantee it. Reliability is guaranteed by other means - i.e. reliable RC control link to prevent the failsafe in the first place; making reliable wiring; using quality GPS modules; validating the configured behavior in a controlled environment. If a pilot chose to YOLO and fly to 20km at maiden - no FC will guarantee a safe return.

Jetrell, 2019-04-06:
_@digitalentity - failsafe_off_delay is ignored when failsafe is executing RTH. If you have proof of the otherwise - please share your data._

I don't have the log for this flight. But I do have DVR footage. 
Here are some screen shots. This was using 2.0.1 firmware.

The first snap is at the moment Timeout occurred and the OSD started to switch to the Stats. It can be seen that it was in RTH from the _systems message_, saying "En route to home"
The second image shows it was disarmed by TIMEOUT. I'm sure you can tell me under what conditions in the code, that will happen.

[image]
[image]

_@digitalentity You seem to misunderstand the purpose of failsafe - to make an attempt to save the aircraft, not guarantee it. Reliability is guaranteed by other means - i.e. reliable RC control link to prevent the failsafe in the first place; making reliable wiring; using quality GPS modules; validating the configured behavior in a controlled environment. If a pilot chose to YOLO and fly to 20km at maiden - no FC will guarantee a safe return._

To say that I misunderstand the purpose of Failsafe, is like you saying that you know how much experience that I have in RC and FPV. 
I have had 1000's of hours of FPV flights. And am well aware of how important hardware preparation and systems ground testing is.
But I do admit...I have had more issues in the last 6 month. From the time I started using iNav, than I ever had in all my years before.
So I thought it would be good to get to the bottom of the issues. And bring them to the developers attention.

So why are you speaking in such a condescending way to me? Without even a word stated about the issue of this post?  

If I had not been this thorough in testing. Do you think that I would have found these problems?
I also found them because I have other flight control systems that don't have these issues; which I can compare iNav too?
Are you interested in getting to the bottom of these issue and making iNav better? Or are you just going to assume I'm a beginner and know nothing. 

I can assure you Konstantin...That iNav may have free software from the input of you guys,
But its far from free...when the user is a beta tester, to iron out bugs. Under these conditions, it doesn't take long to add up to the cost of a proven FC package.
That it not meant to be an insult. Its the truth. 

But if I didn't care to see iNav get better. I would just put ET Vectors on all my models and cut my losses.
So now the ball is in your court. Are you going to look at my findings and try to work out why this is? Or just let it go.....assuming that i'm a troll.

digitalentity, 2019-04-06:
Disarm reason "TIMEOUT" has nothing to do with failsafe. It's triggered be a combination of stick arming, MOTOR_STOP and THROTTLE_LOW lasting more than `auto_disarm_delay` (5 seconds). This was discussed multiple times and is always deemed to be a setup issue, not a code bug. 

To be more specific we need a BB log, but I think this is likely caused by your radio system dropping throttle to low **before** reporting link loss as a failsafe. And yes, if you would have tested failsafe on the ground - you would very likely noticed this disarm before flying.

Regarding the previous issue you reported - I was able to locate a bug in the code that prevents emergency landing from activating during initial RTH climb - fixed by #4582. Thanks for helping figure out this extremely rare edge-case.

In all other BB logs FC behaves as expected. The "dropping throttle to zero" issue is working as it should. Your quad is sitting on the bench, while FC expects it do descent. So FC does the only thing it can to get the quad down - gradually reduce throttle and hoping that the quad will finally start descending.

Ans speaking of the non-free-ness in a sense that user is a beta tester. Please have a look at ET warranty text, especially the part saying that "by the act of setup, use, or assembly, the user accepts
all resulting liability" - meaning exactly that ET gives you no guarantees that the thing will behave as you want it to behave. This is no different from INAV or any other open-source (and 99.999% closed source) products, with the only difference being the price tag of ET Vector.

Since we have found that firmware is behaving as it should in all but one cases (#4582) I'm closing this issue and tagging this as support. Fur further support please join [INAV Telegram channel](https://t.me/INAVFlight).

closed 2019-04-06 by digitalentity

Jetrell, 2019-04-06:
Thank you for your time and amicable reply Konstantin. I know offense can easily be taken to _written_ words; which is not my objective..... But I would rather get to the bottom of problems and help you guys make inav better.

_@digitalentity To be more specific we need a BB log, but I think this is likely caused by your radio system dropping throttle to low before reporting link loss as a failsafe. And yes, if you would have tested failsafe on the ground - you would very likely noticed this disarm before flying._

Firstly. This was not my quad. But rather a guy I fly with. I just told him that I would bring up the matter here.
Although I did run some ground tests to find out for myself what the cause was. And you are correct. It was `auto_disarm_delay` and not `failsafe_off_delay` as I was lead to believe.
In a combination of the RX failsafe throttle position being set to zero (which is a manufacturer recommendation). `MOTOR_STOP` enabled with `auto_disarm_delay`.
Under these settings, when a signal loss failsafe occurs. The radio receiver outputs zero throttle, which then starts the countdown before RTH proceedure is disabled.

**I thank you for the recommendation to remove stick arming. So this misunderstood setting will not cause problems for anyone again.** 

With respect to topic issue. Thank you for looking over the logs. I understand that ground tests will never bring a conclusive result, in this case. 
It's not likely to happen to me again....But I would not like to see it happen to anyone else.
I am grateful that you have opened a PR regarding this :+1: 

Please don't take my statement about the Vector as a cheap shot at iNav. But I was using it as comparison. Because its a flight controller that I know very well. And one that I have tested in every way, over the years.

Thanks again for your professionalism in this matter.

digitalentity, 2019-04-07:
No offense taken @Jetrell.

One major advantage (and disadvantage) of INAV is flexibility. One firmware to suite them all - boards, aircrafts, RC systems, flying scenarios etc. Unfortunately this comes with a lot settings to tweak and a lot of edge cases that are hard to track down, especially when the only thing we have to troubleshoot is a DVR footage.