iNavFlight/inav #4488: MAgnetometer calibration at the field ( Switch assigned)

rcjarvis, 2019-03-10:
## Current Behavior
I flew a different location today, 50 miles away and of course, pos hold was toilet bowling.

## Desired Behavior:
Being able to calibrathe the magnetometer straight from the radio(tx)

## Suggested Solution
Launching a magnetometer calibration either by toggling rth switch or assigning a switch.>

## Who does this impact? Who is this for? Everybody who flies at different locations and can not always have a laptop on hand.
 Could save a day at the field when toilet bowl occurs. Would also be a great part of the checklist for those who often fly at different  remote places.

Thanks for considering.

teckel12, 2019-03-10:
@rcjarvis You can already do this (and many other things). Have you read this documentation?

https://github.com/iNavFlight/inav/blob/master/docs/Controls.md

See the stick positions/functions graphic where calibrate magnatometer is listed.

rcjarvis, 2019-03-14:
HI Tim and thank you.. I have tried this at the field today and I think I get it done because I hear a beep when I do the stick positions for compass cal, I tumble the drone as I would with the Pixracer, all axis, 360+ degrees in all axis ccw.... After some 30 seconds I hear a second beep, everything seems ok... Do i have to do another stick position to enter the data collected by the calibration because I stil have the Fibonnaci flight ( toilet bowl) after many attempts...

Thanks for your time !

https://github.com/iNavFlight/inav/blob/master/docs/Controls.md

teckel12, 2019-03-17:
@rcjarvis The toilet bowl is probably because you don't have your compass oriented correctly. No amount of calibration will fix an incorrectly orientated compass.

What model compass do you have? If it's  BN-880, it's probably 270 Flip or 90 Flip.  Make sure you follow correct procedures for making sure your compass is oriented correctly. Don't even bother calibrating (or even flying IMHO) of you don't have your compass orientation correct.

rcjarvis, 2019-03-17:
HI Tim, Thanks again for your input. I did not introduce myself conveniently, I'm not totally new at flying with GPS, just new at Inav, I have flown most apm's, pixhawks, pixracers, Eagle tree Vectors with success... My F-450 now boasts a Matek f 722Se,  Gps is a Matek with Magnetometer ( 5883) The little white arrow is pointing forward, I did use a 3 inch mast on the top plate to keep GPS and mag away from  the elctro noise and other plausible interferences. I used to run a Pixracer on the same frame and the GPS is installed exactly where the Pixracer's gps was and  working flawlessly. I wanted to try out the Matek instead to bring the costs down for the drone piloting school I work at.... so this is exploratory. I prefered the way it flew (Matek+ Inav) right until I installed the GPS... and we need the GPS and RTH for first level trainees... If Geofence ever becomes available, I will have major chances to swap the costly and unreliable Pixracer for a Matek/Inav !!!

teckel12, 2019-03-17:
@rcjarvis Does Matek list in their manual the orientation you should set in INAV?  Just because it's facing the direction of the arrow doesn't mean that you don't need to set the orientation to flip or even rotated as well. For example, for the BN-880, facing the correct forward direction, the settings are still 270 flip. 

Did you verify on the bench that you could turn the quad and in the configurator it would stop and lock the orientation without drifting?

I turn it to due north, then south, etc. stopping at each 90° and waiting a few seconds to make sure there's no drift. If there is, the configurator orientation is wrong. There's only 8 options, so trying all doesn't take but a few minutes. But guessing that your GPS is from Matek, I'd guess they'd document the correct orientation.

rcjarvis, 2019-03-17:
Hi Tim, again, I’m very greatful for your time, as a matter of fact I was missing 3 things, 1: enable Galileo, 2; flip 270 ( I guess this is done in cli but I did not yet connect to do it ) and mag declination I have to set ( -14.7 where I usually fly) ... I will do the flip thing as soon as I get on my laptop !!! Thx again will let you in on my progress!

teckel12, 2019-03-17:
@rcjarvis Glad to help!

rcjarvis, 2019-03-18:
HI Tim, to set the flip at 270, is it in the line set align_mag = DEFAULT

Or do we have to change either in roll, pitch or Yaw( Yaw seems to make sense) so it would be

set align_mag_yaw = 270

This is what I have done for now ! I'll try tomorrow!

Dan

teckel12, 2019-03-18:
@rcjarvis Probably best to just use the configurator and try each of the options (starting with whatever Matek suggests). There's 4 orientations and 4 more in flip (sensor upside down) for a total od 8 orientation options. All you want to orient is the mag sensor, not the FC orientation.

There's documentation and videos on doing this, but really it's as easy as setting an orientation, testing by pointing different known directions and making sure there's no drift. If there's drift, try the next. There's only 8 so it doesn't take but a couple minutes.

rcjarvis, 2019-03-18:
HI Tim, and everyone wanting to get gps working good... Here goes I'll try this in the afternoon: 
[image]

So I  Switched Galileo on, I finally where to actually do the flip (picture above) I saved and rebooted + I changed set align yaw to 0 and again saved... So I'll keep you all posted on my progress !

teckel12, 2019-03-18:
@rcjarvis You do the testing on the bench, not in the air. You can tell if it's correct in a couple seconds on the bench.

rcjarvis, 2019-03-18:
On the bench it holds way better all positions even when propless motors are spinning ! So I hope for the best!

rcjarvis, 2019-03-20:
WOW... Flies so well... So to me this (flip 270 + galileo) has fixed the issue... Other than the fact that when I activated the retracts (propless on bench) everything went fine... but then in flight as soon as I flicked the retracts switch... the drone flipped and began automutilating... Kill switch did not stop the motors but approx 10 seconds ( seemed forever) and they stopped... Close inspection revealed no damage aside from some light surface scratches on the props... I put some brand new props on, removed the whole retract system and tried another flight... I had a slow wobble on roll and pitch so I will recalibrate and try again today but for the 10 minutes I flew prior to the retract incident.. it flew as good if not better on gps than my dji craps...

So, resolved...thanks Tim !

Dan

closed 2019-06-02 by stale[bot]