iNavFlight/inav #4489: Bypass sat count or turn off GPS altogether if desired

rcjarvis, 2019-03-10:
## Current Behavior
<If flying indoors on some occasions or if gps is not required for a safe flight. GPs count can keep us from enjoying a quick flight. Setting gps count to 1 or 0 would render the gps use unsafe.
## Desired Behavior
Toggling a switch to either bypass sat count or bypass the gps altogether ( bypassing gps would be best)

## Suggested Solution
Making it possible to take off when no sat are available but keeping the gps option when possible/needed

## Who does this impact? Who is this for?
All users using GPS navigation on small quads or planes who enjoyoccasionnal flying indoors
## Additional context
When problems occur, for better troubleshooting etc...

teckel12, 2019-03-10:
@rcjarvis I actually kinda like this. But I would allow the extra arming safety to be turned on or off via a switch. That would resolve the GPS lock as you suggest, but also the angle etc. Basically, flip a switch, arm when you want overriding extra arming safety. I'd use this often.

rcjarvis, 2019-03-11:
Thanks for considering, and thanks Tim for answering my other post... These stick functions are awesome !

malckris, 2019-03-13:
I test/fly/play in my barn when the weather is not cooperating (usually too much wind)... I just change  "nav_extra_arming_safety" for indoor flight... for convenience I put wifi card on the copter so a change is real quick via my phone... controlling this setting with a switch as teckel12 says would be great...

DzikuVx, 2019-04-13:
will be done as part of #4474

rcjarvis, 2019-04-14:
Thanks for considering Pawel !

commit e20790bc (2019-04-16): Add support for emergency arming

Enabling and disabling the arming switch 10 times during a 10s
window will override checks for level, navigation unsafe, compass
not calibrated and some hardware failures (it will only enforce
gyro and acc to be working).

FLASH +224, RAM +24

Fixes #4489 and #4515
commit 1809215d (2019-04-18): Add support for emergency arming

Enabling and disabling the arming switch 10 times during a 10s
window will override checks for level, navigation unsafe, compass
not calibrated and some hardware failures (it will only enforce
gyro and acc to be working).

FLASH +224, RAM +24

Fixes #4489 and #4515
PR #4616 (2019-05-26): Add support for emergency arming
Enabling and disabling the arming switch 10 times during a 10s
window will override checks for level, navigation unsafe, compass
not calibrated and some hardware failures (it will only enforce
gyro and acc to be working).

FLASH +224, RAM +24

Fixes #4489 and #4515
commit ca3f21e7 (2019-05-26): Add support for emergency arming

Enabling and disabling the arming switch 10 times during a 10s
window will override checks for level, navigation unsafe, compass
not calibrated and some hardware failures (it will only enforce
gyro and acc to be working).

FLASH +224, RAM +24

Fixes #4489 and #4515
closed 2019-06-18 by digitalentity