iNavFlight/inav #4496: How To Use The Benewake TFmini

CradyLab, 2019-03-11:
I have a couple of questions about using the Benewake TFmini with iNAV and can't seem to locate any comprehensive documentation.  My setup is a Bixler 3 with a Mateksys F405 WING fc and I'm hoping to use the TFmini for terrain following and landings.

1) Is the TFmini actually supported in iNAV and if so could someone please point me toward step-by-step instructions on the installation and setup
2) If it is supported, which interface is required: UART or I2C 

Thanks!

shellixyz, 2019-03-11:
You just need to connect it to an UART, enable "Rangefinder" as sensor in the ports tab for the right UART then set `rangefinder_hardware` to `BENEWAKE`. It will work like any other rangefinder in iNav. I don't think rangefinder are used for fixed wing models though.

CradyLab, 2019-03-11:
Not working for me =(  I just get a red Sonar icon.  Connections are correct but it isn't being recognized

DzikuVx, 2019-03-14:
I checked today and it is working when connected to free UART

CradyLab, 2019-03-14:
Thanks, Pawel.  I'm trying to test the TFmini using an Arduino to see if it's possibly defective.  Question:  Did you need to use a level shifter?

DzikuVx, 2019-03-14:
No level shifters. Both TFmini and STM32 are 3.3V

CradyLab, 2019-03-14:
Looks like my TFmini is DOA  ='(  Hopefully I'll get a functional replacement

Thanks guys!

DzikuVx, 2019-03-14:
Too bad. Closing then

closed 2019-03-14 by DzikuVx

rparts1, 2020-04-14:
how do i get benewake tf mini to show up on OSD using f722 wing ? have it working but now showing up on OSD

zied923, 2024-08-24:
> I have a couple of questions about using the Benewake TFmini with iNAV and can't seem to locate any comprehensive documentation. My setup is a Bixler 3 with a Mateksys F405 WING fc and I'm hoping to use the TFmini for terrain following and landings.
> 
> 1. Is the TFmini actually supported in iNAV and if so could someone please point me toward step-by-step instructions on the installation and setup
> 2. If it is supported, which interface is required: UART or I2C
> 
> Thait have to be powred  from the battry to work