iNavFlight/inav #4503: Sbus doesn't command motor : Matek F405 wing fcb

pincorr, 2019-03-12:
Hi,

here my current setup kept as simple as possible in order to verify servos and motor:
Inav 2.1.0;
Mini Talon aircraft
matek F405 wing fcb;
Futaba R7008SB receiver connected to the board via Sbus pins;
motor esc wired to ESC-1 pads and S1-G pins;
servos configured  in the Mixer as described in Matek website for Vtail configuration (Mini Talon and similar);
no gps;
no failsafe;

The problem:
control surfaces servos work correctly when activated by transmitter/receiver while motor doesn't work.

Any advice how to solve the problem ?
Thank you

I enclose cli file

[INAV_cli_MINI_TALON_20190312_225518_no_motor.txt](https://github.com/iNavFlight/inav/files/2958937/INAV_cli_MINI_TALON_20190312_225518_no_motor.txt)

DzikuVx, 2019-03-13:
You have to ARM the flight controller for motor to start running.

pincorr, 2019-03-13:
Hi Pawel,
i did it but nothing works. I activated only two modes (ARM and MANUAL) just to reduce the number of switches that I could forget to activate. Ch10 is for ARM but even if activated it seems be ineffective. Did I forget something in the INav setup ? From configurator the motor works but as I move to the transmitter something happen in the fc board that I don't know how to manage.  I have checked all pins and solders just for completness, but they are ok

DzikuVx, 2019-03-13:
https://www.youtube.com/watch?v=7MAdgGkBkXs

pincorr, 2019-03-13:
Pawel, thank you...I saw that video many times :) and i applied all those tips
Nevertheless,  all checks are green but the motor still continues to works only if activated by the configurator. The fact that all servos are working while motor doesn't, still suggest an arming problem but I think I cleared out all those unnecessary setups that should avoid it.  Could be something that can be solved by cli programming ? Some variable that should be OFF instead ON or viceversa ?

shellixyz, 2019-03-13:
@pincorr Check that you can arm. When armed the triangle below the battery indicator will light up yellow. If you can't then post the result of running the status command from the CLI.

[image]

pincorr, 2019-03-13:
hi Michel,
there no way to light up yellow the battery indicator.

I enclose the results of status...hoping that it's what you are asking for.
[image]

shellixyz, 2019-03-13:
Make sure you're flight controller pitch and roll is less than 25° and put the throttle to 0 then you should be able to arm.

Refer to this document for more details: https://github.com/iNavFlight/inav/wiki/%22Something%22-is-disabled----Reasons

pincorr, 2019-03-13:
Board flat and throttle to 0 but motor doesn't spin.

From new status (enclosed) it seems there is nothing wrong compared to previous status.

Question : since I soldered the esc to ESC-1 and ESC-2 is empty, is it possible that signal is inverted , that is, rx is sending signal to ESC-2  rather than to ESC-1 ?

[image]

shellixyz, 2019-03-13:
No the motor signal is output on both. Check that your ESC gets power and that your motor spins with a servo tester.

pincorr, 2019-03-14:
Tested with servo tester connected to G-S1 pins. Signal remains fixed to 997-999 but no input from receiver.

pincorr, 2019-03-14:
Made other test:
- changed  receiver mode from Sbus to Ibus using another tx and rx, connecting signal pin from Sbus to R2.

Signal from G-S1 pins still at minimm and no input from receiver

shellixyz, 2019-03-14:
Servo tester connected to G-S1 pins ?? :scream: 

Is the throttle channel moving in the receiver tab ?

shellixyz, 2019-03-14:
That's not a "servo tester" in my book but ok. I was talking about using a device made to test servos that outputs a PWM signal and connecting it to your ESC to check that your motor/ESC combo was working.

Well if the mixer configured, PWM outputs enabled, manual/angle/acro mode enabled and the board armed, the throttle channel should be forwarded to the motor outputs. Not sure what's going on there...

pincorr, 2019-03-14:
Although I attempted all combinations I just ask you about throttle direction. In order to have signal increasing with stick  going up in the receiver tab, I need to reverse the throttle in the tx. In tx signal is shown opposite to stick direction, then opposite to receiver tab. Which one has the priority ?

pincorr, 2019-03-14:
Got it !!!

Everything as explained and tested up to now but....
I only added a FAILSAFE switch to modes; motor arming light up yellow and motor start to spin.
Then I removed the FAILSAFE switch and , again, the motor start to spin.

In short it seems to work (by now)  but I have lost the logic behind these action. What i understood is that to activate a FAILSAFE switch is mandatory...isn'it ?

shellixyz, 2019-03-14:
No you don't need the FAILSAFE mode on a switch. It is only useful if you want to be able to trigger manually the failsafe procedure from the radio. If you're using S.Bus the failsafe procedure will be triggered automatically if the link between radio and receiver is broken (out of range, radio turned off, ...)

shellixyz, 2019-03-14:
Make sure to follow the fixed wing guide to the letter and everything should work fine: https://github.com/iNavFlight/inav/wiki/Fixed-wing-guide

Please join the [Telegram channel](https://t.me/INAVFlight) or post your question on iNav's Facebook group or RCGroups forum for help

closed 2019-03-14 by shellixyz

pincorr, 2019-03-14:
thank you very much for your support Michel

ciani, 2019-06-01:
I know that this issue is closed, but I have the same problem with no fix.
The variator has not signal from S1 port, regardless in the servo tab all channels works even motor but the f405 not send any signal through S1 to my variator

Any idea about this?

gutgesellt, 2022-04-06:
Hello! 

what exactly is the “arm signal”?!  can it be reproduced without a flight controller? I need the information to unlock my DJI FPV goggles, respectively the Air Unit, to  set the videosignal to 700 mW on an airplane, which I fly without inav... thanks a lot. greetings

csmicfool, 2022-06-23:
> I know that this issue is closed, but I have the same problem with no fix. The variator has not signal from S1 port, regardless in the servo tab all channels works even motor but the f405 not send any signal through S1 to my variator.
> 
> My variator is beeping all time as he does not receiver any signal
> 
> The motor spin with a servo tester but not through the F405

@ciani Did you ever figure out a solution for this?  I'm having the exact same issue.