iNavFlight/inav #4507: Cannot Get Benewake TFmini To Work

CradyLab, 2019-03-15:
I'm on my second Benewake TFmini LiDAR module and I still cannot get iNAV to recognize it when  used with a Mateksys F405 WING

I have set **rangefinder_hardware = BENEWAKE** and have tried UART ports 3 and 5 setting them to RANGEFINDER each time but the SONAR icon in the GUI remains red, indicating the module isn't working

Is there another step I'm missing or something else that needs to be set in the CLI?

shellixyz, 2019-03-15:
Please don't open multiple issues on the same subject. You can still comment on your previous one and reopen it if needed #4496. The only other thing I can think of is check the baud rate. The default is 115200. Did you check that you can communicate with it with the Windows software provided by Benewake ?

CradyLab, 2019-03-15:
The last time I tried to pose a question on a closed subject I was reprimanded so I started a new thread.  

Yes the baud rate is correct.  I finally contacted Benewake directly to seek their help and no, I am unaware of any Windows software available from Benewake

shellixyz, 2019-03-15:
http://en.benewake.com/download

CradyLab, 2019-03-15:
Thanks, I found the link after you suggested it and tried the two TFminis I have via 3.3v FTDI boards but no data received.  Tried swapping the TX/RX lines and set the COM port to 115K

Also tried using the Benewake Updater just to see if I could get any communication but both are giving "no response" errors

Very frustrating

CradyLab, 2019-03-15:
Finally got one of the two TFminis to work with the Benewake test software.  I confirmed the first unit was in fact defective after I re-read the data sheet and realized the FTDI board could not supply enough current.  After wiring in a high capacity power supply the one good unit starting functioning  under the Benewake test program.  I will re-test with the F405 board later today although I have already tested the good unit once already and could not get it to work in iNAV

DzikuVx, 2019-03-15:
Make sure FTmini is powered right from the start. Just pluging USB cable to FC might not (probably does not) power TFmini at all!

CradyLab, 2019-03-15:
Yes, I'm connecting to the battery first, then connecting the USB cable but I'm still getting no response in iNAV.   In the CLI I set rangefinder_hardware = BENEWAKE and I've tried UARTs 3 and 5 - both set to 115K and no success

shellixyz, 2019-03-15:
By default it should be but also make sure your TFmini is not in Pix mode. You can change it from Benewake's GUI.

CradyLab, 2019-03-15:
> By default it should be but also make sure your TFmini is not in Pix mode. You can change it from Benewake's GUI.

Thanks, Michel.  You think it could have shipped in Pix mode?

CradyLab, 2019-03-15:
HUZZAH!  It works!  Apparently it was set to Pix mode at the factory =P

Thanks you so much for the suggestion, Michel and thank you and Pawel for your help =)

closed 2019-03-15 by CradyLab

sensei-hacker, 2024-02-20:
> > http://en.benewake.com/download
> 
> 404 not found

You're clicking a link from five years ago.
They've surely updated their web site in the past five years.

To get to their current download page, you might want to visit the front page of their site, click on the navigation menu, then choose "Downloads" or "Support" or whatever they call the option today.