iNavFlight/inav #4509: When making a non standard quadX mixer, idle motors rev at very high speed

laurentopia, 2019-03-15:
## Current Behavior
When swapping ESC and changing mixer to account for that, idle speeds up as if to compensate for some problem, when I apply some throttle the motors go back to normal idle speed.
(The mixer is fantastic)

## Steps to Reproduce
1. swap left and right ESC
2. make a mixer that swaps yaw and roll values
3. arm

## Expected behavior
idle should be at idle speed

## Suggested solution(s)
Perhaps test this configuration, my guess is there is some old code that hardcode the old 4231 motor order in the quadx configuration

# version
# INAV/MATEKF722SE 2.1.0 Feb 25 2019 / 17:01:07 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

https://pastebin.com/0KYiDsNK

DzikuVx, 2019-03-15:
There is no connection between mixer output and idle (PID output) other than physical. If `Iterm` winds up, this means that you probably made something wrong with the mixer. Even without props it can have enough effect to windup yaw

laurentopia, 2019-03-15:
I think you're right, something is wrong with the mixer I made because when I restored the default quadX mixer, idle becomes smooth again. 
I don't understand though, how can the quad fly with a mixer that causes such result at idle?

This is the default
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

This is what I applied:
mmix 0  1.000  1.000  1.000  1.000
mmix 1  1.000  1.000 -1.000 -1.000
mmix 2  1.000 -1.000  1.000 -1.000
mmix 3  1.000 -1.000 -1.000  1.000

closed 2019-03-26 by DzikuVx

laurentopia, 2019-03-27:
It ended up working well after I switched from DSHOT300 to multishot then back to DSHOT300.
I've noticed oddities with DSHOT before where switching back and forth solved it.