iNavFlight/inav #4510: VTX power : 5 values instead of 4

laurentopia, 2019-03-15:
## Current Behavior
VTX can only go up to power index 4, I have a leaf tx806 which can go up to power index 5 = 1W

## Desired Behavior
VTX can be set to power id = 5 @ 1W

## Suggested Solution
Limit VTX powers to 5 power id instead of 4

## Who does this impact? Who is this for?
anyone with vtx that have 5 power settings

teckel12, 2019-03-16:
@laurentopia What version of INAV and what VTX protocol?  I believe this was resolved in the latest INAV, but protocol information would be required.

laurentopia, 2019-03-16:
@teckel12 2.1 release and smart audio

vicdzen, 2019-04-07:
I want to add here more.
FFPV protocol. 2.4G VTX. FCC firmware has 10, 25, 200, 500, 800. As I know, there is possibility to request min power value from VTX by means of protocol. Also, EU firmware has only 10mW power option, so, please check.

closed 2019-06-20 by stale[bot]

fiam, 2019-08-02:
@laurentopia Tested this a few weeks ago. There’s no way to set the 5th power level in that VTX via SmartAudio. It you tell it to switch to that level, the communications over uart just freeze and you have to restart the VTX.

laurentopia, 2019-08-02:
> @laurentopia Tested this a few weeks ago. There’s no way to set the 5th power level in that VTX via SmartAudio. It you tell it to switch to that level, the communications over uart just freeze and you have to restart the VTX.

Thanks for trying.
It's too bad, the 806 is quite powerful, Do you think it's possible to offset the power range?