iNavFlight/inav #4512: Servo logging breaks Blacbox logs

DzikuVx, 2019-03-16:
@hali9 apparently 6a5e1b18b891a69d7db862d8130cacb695fee4fa breaks blackbox logs if servos are not used, for example on quadx configuration. Logs are written but file is not decoded by both `blackbox_decoder` or Blackbox Explorer.

Not sure what causing it, 6fcea591b477af91177f47b4ff85890bf63f3899 is working fine, 6a5e1b18b891a69d7db862d8130cacb695fee4fa causes 
decoding failure. Example log attached

[LOG00003.TXT](https://github.com/iNavFlight/inav/files/2974152/LOG00003.TXT)

DzikuVx, 2019-03-16:
Interesting, it fails only on one FC: MatekF722, on OmnibusF4 works fine

hali9, 2019-03-16:
I have set the condition always unnecessarily.
should be isMixerUsingServos()

DzikuVx, 2019-03-16:
@hali9 it's not this. I'm experimenting with it for a while. I changed condition, prediction and conditions. And I'm able to reproduce it every time now on every flight controller.

hali9, 2019-03-16:
@DzikuVx I do patch-2 https://github.com/iNavFlight/inav/compare/development...hali9:patch-2
I tested it witch set mmix and without mmix on quadx and it works (I still use SPRacingF3).

DzikuVx, 2019-03-16:
> development...hali9:patch-2

I will test tomorrow then.

DzikuVx, 2019-03-17:
OK, fixed in #4516 Thanks @hali9

PR #4516 (2019-03-17): Fix servo logging into blackbox
Fixes #4512
Replaces #4513

With help of @hali9 from https://github.com/iNavFlight/inav/compare/development...hali9:patch-2 and extra tweaks (predictor is now AVERAGE_2 and correctly use 1500 predictor) it now works fine.
Thanks @hali9 for finding a real cause of the problem!
closed 2019-03-17 by DzikuVx