iNavFlight/inav #4514: OmnibusF7NXT 2.1.0 - Navigation is Unsafe

Rclay11541, 2019-03-16:
## Current Behavior
The firmware update for the OmnibusF7NXT 2.1.0 vs 2.0.1 has an error.
With 2.1.0 OSD & Blackbox now work, however, I'm getting an arm error of **Navigation is Unsafe**. No hadware changes or setting changes, under the old version I didn't get this error.

See issue #4012  [https://github.com/iNavFlight/inav/issues/4012](url)

# version
# INAV/OMNIBUSF7NXT 2.1.0 Feb 25 2019 / 16:57:11 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature THR_VBAT_COMP
feature VBAT
feature TX_PROF_SEL
feature BAT_PROF_AUTOSWITCH
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature BLACKBOX
[... 940 lines cut ...]
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 3500
set battery_capacity_warning = 875
set battery_capacity_critical = 350
set battery_capacity_unit = MAH

---

# version
# INAV/OMNIBUSF7NXT 2.1.0 Feb 25 2019 / 16:57:11 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

shellixyz, 2019-03-17:
Please post the output of the status command run from the CLI

Rclay11541, 2019-03-17:
Now I'm also getting a Run-time calibration failure. If I wait long enough it goes away, attached both Status CLI files below.
[INAV_cli_Long_Range 5 Nav & Runtime.txt](https://github.com/iNavFlight/inav/files/2975791/INAV_cli_Long_Range.5.Nav.Runtime.txt)
[INAV_cli_Long_Range 5 Nav Only.txt](https://github.com/iNavFlight/inav/files/2975792/INAV_cli_Long_Range.5.Nav.Only.txt)

shellixyz, 2019-03-17:
Well if the status command displays only CLI as arming disabled flags it means you can arm once you get out of the CLI. If the gyro runtime cal takes a long time maybe your gyro is noisy (or you're moving the FC). Remember that you need to keep the FC still for something like 10s after powering it up. If you were keeping it still you should check the values reported by the gyro in the sensors tab.

shellixyz, 2019-03-17:
You can refer to this document for the meaning of the arming disabled flags:
https://github.com/iNavFlight/inav/wiki/%22Something%22-is-disabled----Reasons

Rclay11541, 2019-03-17:
Hi, I think you may have prematurely closed the ticket. The reason the ticket was opened is because it arming is being prevented even though there are no flags.

You can see there's a number of I2C errors with the new firmware versus none with the earlier version. The main difference is the SPI was modified between versions to correct the issue with the Blackbox hanging. My thoughts are there may have been unintended consequences during code modification.

shellixyz, 2019-03-18:
What is happening when you are trying to arm?

Rclay11541, 2019-03-18:
Hi, first thanks for helping with this. I really appreciate it!

So after getting a good GPS lock the OSD is flashing "Sensors Calibrating". When i try to arm simply nothing happens. I included the Blackbox logs above. I want able to see anything odd, but I'm a beginner when it comes to code.

Rclay11541, 2019-03-18:
Perhaps the real issue is runtime when in the field? Any thoughts?

Rclay11541, 2019-03-18:
Ah ha! Ok so here's what's going on. Under the old firmware GPS would take forever to get a good lock. With the new firmware it takes less than 30 secs. So I think the long GPS lock time was masking an underlying issue. It takes between 5 & 10 minutes for Runtime to clear...even with very carefully powering up the quad and letting it sit. 

I just powered it up and let it sit for about 10 minutes, the runtime error finally cleared and I was able to arm. The problem is that a **significant amount** of battery life is being used to clear the runtime.

Is this maybe an issue with the Barometer showing up as FAKE?
Attached the blackbox file below.
[blackbox_log_2019-03-17_LONG wait to clear runtime.TXT](https://github.com/iNavFlight/inav/files/2975990/blackbox_log_2019-03-17_LONG.wait.to.clear.runtime.TXT)

Rclay11541, 2019-03-18:
Note there are two Barometers on the OmnibusF7NXT board: MPU6000 and a ICM20608. I don't see either of those listed as options in the CLI.

Rclay11541, 2019-03-18:
see #3697

shellixyz, 2019-03-18:
Try to increase the baro_cal_tolerance setting. Try `set baro_calibration_tolerance = 500`

shellixyz, 2019-03-18:
Indeed the barometer recognized as FAKE is not good. The fake barometer driver is only giving constant pressure and temperature though so it should be an issue for the calibration

Rclay11541, 2019-03-19:
Hi Shellixyz,

Thanks very much, setting to 500 really reduced the barometer calibration time.

Just so I understand, why is it showing as FAKE? What does it mean when it's showing as FAKE? Is it recognizing the barometer at all, or is it just using the GPS?

Thanks again very much for all your help!

closed 2019-03-19 by Rclay11541

shellixyz, 2019-03-19:
I just realized it is displaying FAKE in the configurator but is really LPS25H. It is a bug in the configurator. 500 is probably a bit much for `baro_calibration_tolerance` try to lower it as much as you can while keeping the calibration time acceptable