iNavFlight/inav #4515: Add special arm switch combo to bypass small_angle and unsafe nav checks

shellixyz, 2019-03-16:
The idea is to be able to arm if the model is out of reach (in a tree for example) or just when you know what you are doing and really want to arm.

Alberto suggested switching arm on/off quickly or something like keep switch on for 5 seconds then off for 1 then on.

teckel12, 2019-03-17:
@shellixyz Great idea! I like the quick arm/disarm/arm trigger. Long arm wouldn't work if arming is via switch, no?

Jetrell, 2019-03-17:
Good idea @shellixyz 
Would you also agree that Turtle mode https://www.youtube.com/watch?v=3LL2caB3r88 
could also be a variant along with this, now that inav has Dshot?

laurentopia, 2019-03-17:
A Savior!
Someone suggested another way to trigger bypass-arm with 5 or 10 successive arm-disarm. That one won't be done by the user by mistake and it is easy to remember and perform. 
I wonder if it could happen as a result of an R9 glitch, though.

DzikuVx, 2019-03-17:
This is another case for logic conditions and global functions I'm
currently working on.

shellixyz, 2019-06-28:
Done in 2.2.0, closing

closed 2019-06-28 by shellixyz