iNavFlight/inav #4518: Why i cant arm?

V3j3ta, 2019-03-17:
status
System Uptime: 28 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 15.16V (4S battery - OK)
CPU Clock=216MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: Startup failed
Stack size: 6144, Stack address: 0x20010000, Heap available: 1664
I2C Errors: 45, config size: 4671, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 2, cycle time: 2009, PID rate: 497, RX rate: 110, System rate: 9
Arming disabled flags: ARMSW THR CLI SETTINGFAIL
Invalid setting: nav_mc_braking_speed_threshold

# status
System Uptime: 93 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 15.15V (4S battery - OK)
CPU Clock=216MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: Startup failed
Stack size: 6144, Stack address: 0x20010000, Heap available: 1664
I2C Errors: 45, config size: 4671, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 2, cycle time: 2000, PID rate: 500, RX rate: 110, System rate: 9
Arming disabled flags: THR CLI SETTINGFAIL
Invalid setting: nav_mc_braking_speed_threshold

stronnag, 2019-03-17:
Explained on the wiki

https://github.com/iNavFlight/inav/wiki/%22Something%22-is-disabled----Reasons

V3j3ta, 2019-03-17:
Ok, but i can’t understand the failure and how to solve it. 
It’s my first build.

shellixyz, 2019-03-17:
In the CLI run the `defaults` command and configure your FC again. One of the setting is corrupted `nav_mc_braking_speed_threshold` most likely because you flashed your FC without the "full chip erase" option turned on.

Please join the Telegram channel or post your questions on iNav's Facebook group or RCGroups forum for help. Github issues are for bug reports and feature requests.

closed 2019-03-17 by shellixyz