iNavFlight/inav #4519: Too low speed RTH on Fail Safe

gabrielaltea, 2019-03-18:
Hello, thanks for you hard work on INAV
I had a problem with the RTH in FS. condition

In the first incident, flying between mountains, I lost 'LOS' and video signal and control.
At that moment I touched the transmitter 'RTH' switch immediately.
The quad has entered FS but it has not returned at the adjusted speed (35 km/h), it has done it at only 4 or 5 Km / h. Entering and leaving FS condition?, but in almost 4 minutes, it has only traveled 300 meters.
Then when recovering video I took control and returned home manually. (Video link later)

In the second incident it was similar, but this time I turned off the transmitter, to avoid entering and leaving 'FS' intermittently, on this occasion the quad also made FS / RTH at 4 or 5 km / h and it took almost 9 minutes to travel 500 meters, until the battery runs out.
At this point, I cannot rtecover the Quad, but I know exactly where it is)

I have done some tests after the first incident, I have flown 1 kilometer and pressed the transmitter RTH switch, the quad has returned quickly at 35 Km / h. 
I also took the quad to 500 meters and turned off the transmitter, and the quad has turned quickly and has also returned home at 35 km / h.
I can not find the error in the configuration. The Modes activated where, ALTHOLH, POSHOLD, TURNASSIST, AIR, ANGLE
I need to know if there are an incorrect parameter, to avoid issues with a new Quad.

This is the config I have used:
diff all
version
INAV/MATEKF722 2.1.0 Feb 10 2019 / 16:41:25 (0d3d0ff30)
GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
reset configuration to default settings
defaults noreboot

resources
mixer
mmix 0 1.000 1.000 1.000 1.000
mmix 1 1.000 1.000 -1.000 -1.000
mmix 2 1.000 -1.000 1.000 -1.000
mmix 3 1.000 -1.000 -1.000 1.000

servo mix
servo
feature
feature MOTOR_STOP
feature GPS
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

beeper
beeper -RUNTIME_CALIBRATION
beeper -RX_LOST_LANDING
beeper -BAT_CRIT_LOW
beeper -GPS_STATUS
beeper -ACTION_SUCCESS
beeper -ACTION_FAIL
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -LAUNCH_MODE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

map
serial
serial 4 2 115200 57600 0 115200

led
color
mode_color
aux
aux 0 0 0 1300 2100
aux 1 1 0 1300 2100
aux 2 26 1 925 2100
aux 3 4 1 1700 2100
aux 4 3 1 1300 2100
aux 5 9 1 1700 2100
aux 6 8 2 1525 2100
aux 7 11 5 1400 2100

adjrange
adjrange 0 0 8 900 2100 12 8

rxrange
temp_sensor
osd_layout
osd_layout 0 1 12 0 H
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 6 22 11 V
osd_layout 0 9 1 9 H
osd_layout 0 13 23 1 V
osd_layout 0 15 12 5 V
osd_layout 0 20 18 12 V
osd_layout 0 21 0 12 V
osd_layout 0 22 14 11 V
osd_layout 0 23 2 0 V
osd_layout 0 24 24 5 V
osd_layout 0 25 22 5 V
osd_layout 0 26 23 2 V
osd_layout 0 28 23 11 H
osd_layout 0 30 2 10 V
osd_layout 0 32 11 0 V
osd_layout 0 38 12 4 V
osd_layout 0 40 1 2 V
osd_layout 0 41 1 8 V
osd_layout 0 42 1 9 V
osd_layout 0 46 12 9 H
osd_layout 0 54 11 1 V
osd_layout 0 88 25 4 H

master
set looptime = 500
set align_gyro = CW270
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_stage2_lowpass_hz = 145
set align_acc = CW270
set acc_hardware = MPU6500
set acczero_x = 4
set acczero_y = -19
set acczero_z = -10
set accgain_x = 4091
set accgain_y = 4093
set accgain_z = 4031
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set mag_declination = -3800
set magzero_x = 10
set magzero_y = -114
set magzero_z = -49
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 16
set blackbox_rate_denom = 5
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 4
set failsafe_procedure = RTH
set current_meter_scale = 240
set bat_voltage_src = SAG_COMP
set model_preview_type = 3
set alt_hold_deadband = 80
set inav_reset_altitude = EACH_ARM
set nav_use_midthr_for_althold = ON
set nav_auto_speed = 1000
set nav_manual_climb_rate = 500
set nav_min_rth_distance = 200
set nav_rth_climb_ignore_emerg = ON
set nav_rth_alt_mode = EXTRA
set nav_rth_altitude = 600
set nav_mc_bank_angle = 26
set nav_mc_braking_bank_angle = 31
set osd_video_system = NTSC
set osd_time_alarm = 16
set osd_alt_alarm = 200
set osd_dist_alarm = 1300
set osd_left_sidebar_scroll = ALTITUDE
set osd_main_voltage_decimals = 2
set name = LaPolla

profile
profile 1

set mc_p_pitch = 45
set mc_i_pitch = 28
set mc_d_pitch = 40
set mc_i_roll = 28
set mc_d_roll = 35
set mc_d_yaw = 15
set max_angle_inclination_rll = 300
set max_angle_inclination_pit = 310
set dterm_lpf_hz = 75
set rc_yaw_expo = 70
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 50

battery_profile
battery_profile 1

set vbat_warning_cell_voltage = 360

battery_profile
battery_profile 2

battery_profile
battery_profile 3

restore original profile selection
profile 1
battery_profile 1

save configuration
save

In this short video you can see the image of the OSD and the camera during those 4 minutes.
At the beginning and at the end there is an image of FPV, in the middle you can only see the image of the camera, because of the loss of signal.
https://youtu.be/rDCMPzwVZZ0

Thanks in advance for the help.

teckel12, 2019-03-18:
@gabrielaltea Maybe increasing `nav_mc_bank_angle`? 26 degrees seems like it would be enough, but maybe not enough to travel at the desired 10m/sec or maybe lowering the speed from 1000 to 500 or increasing the max bank angle.

gabrielaltea, 2019-03-18:
> @gabrielaltea Maybe increasing `nav_mc_bank_angle`? 26 degrees seems like it would be enough, but maybe not enough to travel at the desired 10m/sec or maybe lowering the speed from 1000 to 500 or increasing the max bank angle.

Thanks as you can see in the short video, the bank angle is about 0.1 and 1 (a the bottom left,, above the satellite count)
With a bank of 18 the Quad reaches 54 Km/h at the end of the video.
And 31 Km/h with 9 degrees
The banking is not the problem, almost without strong wind

Thanks enyway

gabrielaltea, 2019-03-18:
Watching the video again, the speed of 5 km / h seems to be that of the wind that pushes the Quad
Since the Pitch angle is close to 0 degrees
In that case, even though the phrase "returning home" is seen on the screen, the quad is actually stopped almost all the time.

teckel12, 2019-03-18:
@gabrielaltea Sorry, on my phone so I didn't watch the video. Is the mag declination of -3800 correct? Looking at your settings compared to mine to find differences that could be causing this.

giacomo892, 2019-03-18:
@teckel12 Manual declination is overwritten by the FC since automatic declination is on.
@gabrielaltea any BB logs?

gabrielaltea, 2019-03-18:
TThe declination of 3800 is correct for my position, anyway I did not know that the automatic configuration overwrites this. The first time it happened, I did not have BB activated, the second time I had it active, but the Quad was lost. The manually activated RTH worked perfectly, landing 1 meter or less from the point of origin, the compass also worked well always, the Beitian 880 was located on a 12 cm mast.
The 'FS' also worked well, in the artificial test, turn off the transmitter, and the Quad came home from 500 meters at a maximum speed of 35 km / h, in a straight line, without jumps or changes of course.
That is what puzzles me the most, despite being using version 2.1.0
E tests behaves well, but in a real failure of signal loss, has not returned.
4 minu8tos in the site without moving, I regained control by changing my position and raise my height a little and stretch my arms, that's why in a moment, at the end of the video, I recover control and return manually.
At the same time, before landing, it is that I move it 500 meters away from a sonocida position and turn off the transmitter, to test, and surprisingly, it returns home and even lands alone, without failures.

tiriad, 2019-03-19:
From my humble knowledge I would try the following,
set align_gyro = DEFAULT
set align_acc = DEFAULT

Arbitrary board rotation in deci-degrees (0.1 degree), to allow mounting rotated 
set align_board_yaw = 270

The value of the magnetic declination seems to me very high, I would put it to 0 as default because it is automatic with the gps, or if you insist insert the value that gives this page http://www.magnetic-declination.com/

wiki https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup

> Default DJI GPS puck pointing forward is set with CW180FLIP, but can be changed with CW0FLIP, CW90FLIP, CW180FLIP or CW270FLIP
F3 based board and newer uses default automatic magnetic declination, if your on F1 board or want to change magnetic declination manually you have to set correct declination of your spesific location, which can be found here: www.magnetic-declination.com. If your magnetic declination readings are e.g. +3° 34' , the value entered in the iNav configurator is 3.34 (3,34 in some locales). In the CLI, the same effect would be set mag_declination = 334. For west declination, use a minus value, e.g. for 1° 32' W, set mag_declination = -132. In all cases (both CLI and GUI), the least significant digits are minutes, not decimal degrees.
Calibrate your compass according to compass calibration
Note to change magnetic declination manually on F3 or newer board you have to turn off automatic function. set inav_auto_mag_decl = OFF.

gabrielaltea, 2019-03-19:
> From my humble knowledge I would try the following,
> set align_gyro = DEFAULT
> set align_acc = DEFAULT
>
> Arbitrary board rotation in deci-degrees (0.1 degree), to allow mounting rotated
> set [...]

I'll set the magnetic declination to 0, the FC is a "no brand" F7 (722) The set align_gyro, align_acc and align_mag have the values that make the model and the Quad pointing and moving in the right direction, the Beitian 880 has the cable The connector points back, and the compass has been verified with an analog compass.
In flight the compass points to the right direction, but I will try with those values anyway.
What makes my crazy is that the RTH works perfectly on every test I did, from 100m to 2.2km (video signal lost at that distance). The OSD arrow points to the house perfectly, without jumps.
And also the fact of being in the air without moving, pointing in the right direction,, ostraging the arrow and the distance to the right house, as seen in that video capture.
The Quad was for almost 4 minutes like that, until I managed to change position and recover LOS
[image]

tiriad, 2019-03-19:
> the FC is a "no brand" F7 (722) 

I thought the fc was rotated. If the arrow of the fc is towards the front leaves the values that you had.
It would be useful a picture of how you have mounted it on the quad and make sure that the version you have loaded corresponds to your model

gabrielaltea, 2019-03-19:
> > the FC is a "no brand" F7 (722)
> 
> I thought the fc was rotated. If the arrow of the fc is towards the front leaves the values that you had.
> It would be useful a picture of how you have mounted it on the quad and make sure that the version you have loaded corresponds to your model and install clean the latest stable version
> 
> You have played with the mixer of the motors?
> This is the one I have by default
> 
> > mmix 0 1.000 -1.000 1.000 -1.000
> > mmix 1 1.000 -1.000 -1.000 1.000
> > mmix 2 1.000 1.000 1.000 1.000
> > mmix 3 1.000 1.000 -1.000 -1.000

yes, I changed the configuration of the motors, since the output of the connector of the board and the ESC 4in1 did not have the required order, that is not cause of problems if it is well done.
This is the controller and the plate 4in1, the arrow on the plate does not indicate the actual orientation, it is surely poorly printed.
I have configured the board as Matek722 since it has the same devices.
It has not given problems or errors.
[image]
[image]

tiriad, 2019-03-19:
It can be useful to order the wires from the connector of the 4 in 1 ESC helped with a pin and not have to change the mixer, otherwise I do not know that it can go wrong hopefully someone else can help, you make sure that this accelerometer is not the ICM20689 instead of the MPU6500

gabrielaltea, 2019-03-19:
> It can be useful to order the wires from the connector of the 4 in 1 ESC helped with a pin and not have to change the mixer, otherwise I do not know that it can go wrong hopefully someone else can help, you make sure that this accelerometer is not the ICM20689 instead of the MPU6500

You are right, the accelerometer that I have set is the MPU6500 and according to the data of the plate should be the ICM20689
I do not know if these data are correct, I will look at the plate to try to identify the correct one.
I do not know if they are also compatible in the driver, since the values ​​and the calibration were always correct, as well as the angle of the Quad at all times, except in a real FS.
It is possible that you have the key to the topic.
I'm going to verify

closed 2019-06-01 by stale[bot]

craigmunday, 2020-07-25:
> What makes my crazy is that the RTH works perfectly on every test I did, from 100m to 2.2km (video signal lost at that distance). The OSD arrow points to the house perfectly, without jumps.
> And also [...]

I believe I experienced a similar issue recently when Failsafe RTH was triggered from a loss of radio link.  I had tested both RTH and failsafe RTH (switching the transmitter off), and both were working flawlessly on every occasion.    I even tested RTH on a high wind day (25 km/h) and the quad made it back.

However on a recent occasion when failsafe RTH was triggered due to a loss of radio link the quad ended up pointing home while hovering level and with no apparent attempt to fly home. Actually the quad had a drift sideways (I think my alignment is out) and eventually emergency landed (crashed) because I assume it hit the abort threshold when the distance from the FS trigger point increased to 500m.  This was a light wind day.

I wonder whether this is a bug given there are a couple of examples, but I am not ruling out operator error.  I am running iNav 2.5.1.