iNavFlight/inav #4520: SPRACINGF3 No configuration received within 10 seconds, communication failed

nikola-v, 2019-03-18:
## Current Behavior
Cannot connect to Spracing F3 Deluxe.
No configuration received within 10 seconds, communication failed

## Steps to Reproduce
1. Put SPRACINGF3 to boot mode
2. Load flash SPRACINGF3 2.1.0.
3. Reboot SPRACINGF3  (boot pin is not connected)
4.  Open INAV configurator 2.1.4 and open connection

## Expected behavior
Configurator should open configuration for SPRACINGF3 but it just says:

2019-03-18 @ 19:56:24 -- Serial port successfully opened with ID: 74
2019-03-18 @ 19:56:34 -- No configuration received within 10 seconds, communication failed

## Suggested solution(s)
None

- FC Board name and vendor: SPRACINGF3 Deluxe
- INAV version string: Configurator 2.1.4

shellixyz, 2019-03-19:
What OS are you running the configurator on ?

nikola-v, 2019-03-19:
Windows 10 Pro
Chrome Version 72.0.3626.121 (Official Build) (64-bit)

shellixyz, 2019-03-19:
Does it work with an older version of the firmware and configurator ? For example FW 1.9.1 and configurator 1.9.3: https://github.com/iNavFlight/inav-configurator/releases/download/1.9.3/INAV-Configurator_win32_1.9.3.zip

shellixyz, 2019-03-19:
Also please use the standalone version of the configurator. Chrome version are not supported anymore.

nikola-v, 2019-03-20:
Standalone configurator doesn't work, it opens up with black screen and stays that.
I have tried to run it as administrator but still it is black.
I have tried with latest cleanflight / cleanflight configurator and it works without issues.
Now going to try with INAV 191/193 as you proposed.

closed 2019-06-02 by stale[bot]