iNavFlight/inav #4522: sp racing f3 works but not connect pc

oguzhantasdemir, 2019-03-19:
my sp racing f3 work with last setting. no connect pc and not install new fw. i tried many method. but not connect. only connect via debug port(swd). i think my uart1 inverter is broken. i have an idea. if make msp in uart2, i can connect to pc with ftdi adapter via uart2 port. and sorry my bad English 😅

closed 2019-03-19 by shellixyz

oguzhantasdemir, 2019-03-19:
why closed my issue?

shellixyz, 2019-03-19:
Why not ?

oguzhantasdemir, 2019-03-19:
I have a problem. is not it?

shellixyz, 2019-03-19:
Is your issue a bug or feature request ?

oguzhantasdemir, 2019-03-19:
not both. please read my first article.

shellixyz, 2019-03-19:
Then sorry but it is not the right place to post about your problems. Github "issues" are for reporting bugs and asking for new features. This is not a support forum.

oguzhantasdemir, 2019-03-19:
i am sorry. i didnt now. thank you so much.

shellixyz, 2019-03-19:
No problem. Please join [iNav's Telegram channel](https://t.me/INAVFlight) if you want live help.