iNavFlight/inav #4524: IST8310 compass does not work any more with inav 2.1.0

Ralfde, 2019-03-20:
**The IST8310 does not work any more with inav 2.1.0 and Omnibus F4PROV3 board?
Adress of compass is set to 0x0C

The MS5611 on our GPS #33602 works with I2C, so the problem seem to be not the I2C hardware of the processor. It also works with OmnibusF7NXT 2.0.1**

# diff all

# version
# INAV/OMNIBUSF4V3 2.1.0 Mar 19 2019 / 12:44:58 (9171f3a60)
# GCC-7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]

# reset configuration to default settings
defaults noreboot

# resources

# mixer
mmix 0 1.000 0.000 0.000 0.000

# servo mix
smix 0 2 0 -100 0
smix 1 3 0 -100 0
smix 2 4 1 -100 0
smix 3 5 1 100 0
smix 4 4 2 -100 0
smix 5 5 2 -100 0
smix 6 2 11 100 0
smix 7 3 11 -100 0
smix 8 4 11 40 0
smix 9 5 11 -40 0

# servo
servo 2 1070 1730 1500 -100
servo 3 1270 1900 1500 -100
servo 4 1100 1900 1500 -100
servo 5 1250 1750 1500 100

# feature
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 20 1 115200 115200 0 115200
serial 0 64 115200 115200 0 115200
serial 5 2 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 5 1800 2000
aux 1 1 4 1000 1200
aux 2 3 4 1000 1200
aux 3 9 6 1400 1600
aux 4 8 6 1000 1200
aux 5 10 4 1800 2000
aux 6 27 7 1000 1200
aux 7 28 7 1800 2000

# adjrange

# rxrange

# temp_sensor

# osd_layout
osd_layout 0 0 22 2 V
osd_layout 0 1 15 2 V
osd_layout 0 2 0 0 V
osd_layout 0 9 1 2 H
osd_layout 0 11 2 2 V
osd_layout 0 12 9 2 V
osd_layout 0 13 2 10 V
osd_layout 0 14 2 11 V
osd_layout 0 15 23 7 V
osd_layout 0 22 14 8 V
osd_layout 0 23 2 9 V

# master
set looptime = 2000
set gyro_hardware_lpf = 188HZ
set acc_hardware = MPU6000
set acc_lpf_hz = 5
set acczero_x = 69
set acczero_y = -8
set acczero_z = -81
set accgain_x = 4103
set accgain_y = 4100
set accgain_z = 4049
set align_mag = CW270
set mag_hardware = IST8310
set mag_declination = 240
set baro_hardware = MS5611
set pitot_hardware = NONE
set receiver_type = SERIAL
set rssi_channel = 16
set serialrx_provider = SUMD
set min_throttle = 1000
set max_throttle = 1900
set motor_pwm_rate = 50
set failsafe_procedure = RTH
set platform_type = AIRPLANE
set model_preview_type = 28
set nav_rth_altitude = 2000

# profile
profile 1

set mc_p_pitch = 48
set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_p_roll = 36
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 70
set fw_p_pitch = 20
set fw_i_pitch = 30
set fw_ff_pitch = 15
set fw_p_roll = 20
set fw_i_roll = 30
set fw_ff_roll = 15
set fw_p_yaw = 45
set fw_i_yaw = 5
set fw_ff_yaw = 15
set max_angle_inclination_rll = 600
set tpa_rate = 50
set pitch_rate = 15
set yaw_rate = 9

# profile
profile 2

set fw_p_pitch = 10
set fw_p_roll = 10
set fw_p_yaw = 10
set max_angle_inclination_rll = 600

# profile
profile 3

# battery_profile
battery_profile 1

set vbat_cell_detect_voltage = 450
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 300
set battery_capacity = 1200
set battery_capacity_warning = 960
set battery_capacity_critical = 1080

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 2
battery_profile 1

# save configuration
save 

**With Omnibus F7NXT 2.1.0 the IST8310 and MS5611 does not work on I2C.
With 2.0.0 and 2.0.1the IST8310 does work, but not the MS5611 on I2C.**

# diff all

# version
# INAV/OMNIBUSF7NXT 2.0.1 Oct 31 2018 / 19:22:00 (a04e68aaa)

# reset configuration to default settings
defaults noreboot

# resources

# mixer

# servo mix

# servo

# feature
feature GPS

# beeper

# map

# serial
serial 0 0 115200 38400 0 115200
serial 1 64 115200 38400 0 115200
serial 3 8 115200 38400 19200 115200
serial 5 2 115200 115200 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# osd_layout

# master
set acc_hardware = MPU6000
set mag_hardware = IST8310
set mag_declination = 240
set baro_hardware = LPS25H
set pitot_hardware = ADC
set serialrx_provider = SUMD

# profile
profile 1

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

digitalentity, 2019-03-20:
Should be fixed by #4523

Ralfde, 2019-03-21:
Thanks a lot, works fine, now. Could you please check also why the MS5611 doesn't work on I2C of the OmnibusF7NXT. It works on OmnibusF4PROV3.

shellixyz, 2019-03-21:
Nice. For the baro MS5611 isn't enabled. The OmnibusF7NXT is supposed to have an onboard LPS25H baro why don't you use the on-board one ?

Ralfde, 2019-03-21:
I just thought, the MS5611 should be available, too. But if it is not activated, then I will use the LPS25H.
How is quality of the LPS25H compare to the MS5611? Is it about the same? I never used it before.

shellixyz, 2019-03-21:
No idea. I never used one. But it should be good enough. Most of the FCs I use have a BMP280 and it seems to be good enough.

closed 2019-03-21 by shellixyz