iNavFlight/inav #4525: Pit mode -> arm  -> power level 2 or 3 ->  disarm -> pit mode

romko1, 2019-03-20:
Hello,
I have  VTX  AKK FX3, FC MatekF405 mini.
At the moment it works for me as follows: I set up the OSD pit mode (OP MODEL = RACE) - restart the FC, it's still pit mode. If I do ARM it goes to the set VTX transmit power but then when I do disarm it just decreases the power to 1 (transmiter power is 25mW, not pit mode = 0.01mW).

I use inavflight where I set:
set vtx_halfduplex = ON
set vtx_band = 5    //race
set vtx_channel = 3
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5732
set vtx_pit_mode_freq = 5732

When I first do the ARM it switches to the desired transmit power. 
However, if I do a disarm it will not go into P (pit mode) but will decrease the power to 25mW (flashed 1). 
When I look through the OSD to the VTX AKK FX3 so the pit mode is deactivated - the OP model is FREE again.

I want to make VTX go to pit mode after disarm -> P flashed.
Can you please tell me how to do this?

Thank you.

BradB111, 2019-03-20:
I believe this is a limitation to the Smart audio protocol for Smart Audio <=2.0 

The new TBS Vtxs support this with Smart Audio 2.1

Hope this helps.

romko1, 2019-03-20:
So is it possible that VTX AKK FX3 only supports smartaudio protocol <=2.0?

BradB111, 2019-03-20:
AKK as far as I'm aware don't use an official implementation of Smart Audio (the reason for the AKK patch on betaflight for example) 
some basic info on this can be found here: https://github.com/betaflight/betaflight/issues/5086

romko1, 2019-03-20:
yes that applies to betaflight, but I didn't find some patch for inavflight to get back to pit mode via disarm.
I currently have inav v.2.0.1 installed on that quadcopter..

BradB111, 2019-03-20:
> yes that applies to betaflight, but I didn't find some patch for inavflight to get back to pit mode via disarm.
> I currently have inav v.2.0.1 installed on that quadcopter..

What i'm saying is its not even 'smart audio' its a vtx that happens to accept old smart audio commands so i wouldn't expect it to work any time soon (if at all). 

If your smart audio works to change channel then iNav either; 
1. has the patch already or 
2. iNav isnt fully compliant with the official TBS standard (using the old implementation). 
(maybe someone else can clarify this i haven't been following iNav for that long)

Putting the vtx into pit mode doesn't work on (as far as I know):

- TBS Vtxs running Smart Audio <=2.0
- other Vtxs running Smart Audio <=2.0
- AKK Vtxs patch
- AKK Vtxs using the old smart audio implementation

The Only Vtxs that will support what you are looking for (putting the Vtx back into pit mode after being taken out of pit mode) are these:
- https://www.team-blacksheep.com/products/prod:unify_pro_hv_race2_m
- https://www.team-blacksheep.com/products/prod:tbs_unify_evo
- https://www.team-blacksheep.com/products/prod:unifypro32_nano
- Other Smart audio 2.1 Vtxs although I'm not aware of any.

romko1, 2019-03-20:
Oh, I understand how you thought it was. I have a new vtx and I thought the pitmode transition from it to it will work.

romko1, 2019-03-21:
I wrote to support AKK and their response: The case is not related with the vtx.

BradB111, 2019-03-21:
You need one of those vtxs I linked above, and you need to make sure smart audio 2.1 feature set is supported on iNav and I don't think it is currently. 

AKK DOES NOT support the official smart audio protocol. Certainly not 2.1 which is required for your use case.

closed 2019-06-03 by stale[bot]

padcom, 2019-06-12:
I just bought the PRO32 Nano VTX, configured as per the example stated in the first comment and PIT mode is _NOT_ automatically engaged when disarming. INAV 2.2.0 RC-2 on Matek F405-WING (MATEKF405SE target). Any idea how to get it to work? Or how to engage the pit mode manually?