iNavFlight/inav #4528: Simple Follow me mode possible?

tott, 2019-03-20:
I'm just getting started in flying and i'm still a terrible pilot but I'm a sufficient coder and know electronics so i'm wondering if there is a way to feed target gps coordinates and heading mid-flight through UART or other serial communication into the flight controller/inav?

If that would be possible I could just drop a 868 Mhz long range capable Arduino on my quad and one in my backpack and transfer the coordinates / heading from the module that I keep with me into the quad and inav. Surely it would need a bit more coding to predict movements and speed and the like but generally this could be a very simple implementation without much effort on the inav side.

It would likely not even require a dedicated GPS to transfer the coordinates of the person as you could simply run a JS / Node script on the Arduino of the carrier and open a website on this device that would read the phones GPS position to update it so hardware costs for such an approach could be <20$.

Would love to hear if there is a way to feed in the necessary values during the flight through some serial channel already. I've seen there are plenty of commands in the CLI but I did not really grasp from the documentation if these would work mid-flight and could be send through other UARTS than the USB connection

shellixyz, 2019-03-20:
There is a way. You can it do through an UART with the MSP protocol. Take a look at the MSP_SET_WP message. With it you can set a virtual waypoint, waypoint 255, setting the immediate destination / desired position.

https://github.com/iNavFlight/inav/blob/master/src/main/fc/fc_msp.c#L2278

tott, 2019-03-20:
This is great. Thanks for the quick answer :+1:

tott, 2019-03-20:
For anyone else looking for this here is additional information about this https://github.com/iNavFlight/inav/wiki/MSP-Navigation-Messages#msp_wp--msp_set_wp

closed 2019-03-20 by tott

jackDonati, 2020-07-29:
Hi guys, can someone explain me how does WP #255 works? MSP navigation message page say that P1 may be used to control heading, but what if I want to set speed [cm/s] used to reach ''follow me'' WP? NAV WP must be engaged together with GCS NAV, POSHOLD and MAG?

mrnams, 2021-10-20:
@tott Though I know coding and electronics, its unfortunate I am not getting time to look into code you mentioned, so considering user perspective what are simple steps to achieve follow me,its better for all who looking for if you provide some document like what hardware required,how to connect them(wiring) and how to configure it in iNav configuration.
Thanks in advance.