iNavFlight/inav #4531: Matek f411 wing softserial problem

joeclone, 2019-03-23:
I need to use softserial for mavlink telemetry but I think because SS for this target is setup for halfduplex it does not work for my needs. I've tried flashing betaflight target matekf411 setup, resource for SS and ran tlm_halfduplex=off to test this and I can get mavlink to work.
Is it possible inav dev to make a matekf411 target with softserial halfduplex=off?

shellixyz, 2019-03-24:
There is no half duplex setting for telemetry for now. It will probably appear in a future version.

joeclone, 2019-03-24:
Is it possible to make it work if we compile a new build for this target with the softserial rx assigned to another pin? In the target.h file...

#define SOFTSERIAL_1_TX_PIN     PA0 // ST1 pad
#define SOFTSERIAL_1_RX_PIN     PA0

shellixyz, 2019-03-24:
Maybe. You can try. Don't forget to define a timer in target.c for the new pin you're going to use for RX.

joeclone, 2019-03-24:
I'm no coder, tried to compile a build but I'm stuck at usr/bin/sh: ruby: command not found.

shellixyz, 2019-03-25:
You can find docs to help you set up a build environment here: https://github.com/iNavFlight/inav/tree/master/docs/development

joeclone, 2019-03-28:
DEF_TIM(TIM1, CH1, PA8,  TIM_USE_ANY,   0, 0), //softserial_tx2 - 2812LED TIM_USE_LED
    DEF_TIM(TIM5, CH1, PA0,  TIM_USE_PPM,   0, 0), //use rssi pad for PPM/softserial_tx1

I'm planning to use PA8 for softserial RX1, which line should I change to reflect this? Would this be correct?
DEF_TIM(TIM5, CH1, PA8,  TIM_USE_ANY,   0, 0), //softserial_tx2 - 2812LED TIM_USE_LED
    DEF_TIM(TIM5, CH1, PA0,  TIM_USE_PPM,   0, 0), //use rssi pad for PPM/softserial_tx1

shellixyz, 2019-03-28:
All you need to do is: `#define SOFTSERIAL_1_RX_PIN PA8`

joeclone, 2019-03-28:
No need to make changes to timers?

shellixyz, 2019-03-28:
No since the timer defined for PA8 (TIM1) is already independent of all the other used timers

joeclone, 2019-03-29:
After spending a whole week messing with getting Inav compiled finally got it working. Tried all modes of compiling on Windows none worked, had to settle on a steep learning curve with Ubuntu.
Thanks to @shellixyz for his help and pointers, finally got this softserial fullduplex working nicely on MatekF411 board doing Mavlink telemetry to Qlrs lora system.

@shellixyz  if I do this, would it still work?
#define SOFTSERIAL_1_RX_PIN NONE

shellixyz, 2019-03-29:
Not sure. I think setting this to none is the same as setting both RX and TX to the same pin

joeclone, 2019-03-31:
Indeed setting it to none it falls back to halfduplex. So as of now I'm stuck losing an output for fullduplex telemetry. Will have to wait for code revision for tlm_halfduplex=off option for inav.

closed 2019-03-31 by joeclone

SnappyFPV, 2019-04-13:
@joeclone how did you solve this and which pins did you use ... plan to use the inav ESP32 LORA on this soft serial ...

SnappyFPV, 2019-04-13:
Changed the SFTSRL2 Version to use S5 and S6 for Serial2 - had test this with GPS - works fine!

#if defined(MATEKF411_SFTSRL2)
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_TX_PIN PA0 // ST1 pad
#define SOFTSERIAL_1_RX_PIN PA0
#define USE_SOFTSERIAL2
#define SOFTSERIAL_2_TX_PIN PB3 // S5
#define SOFTSERIAL_2_RX_PIN PB10 // S6
#define SERIAL_PORT_COUNT 5

Fetzii99, 2020-05-27:
@SnappyFPV Hey! I really like to use a gps in s5/6, can you send me your Firmware for the FC? I don't know how to edit this !
Huge thanks!