iNavFlight/inav #4532: Servo Auto Trim - unexpected behaviour ?

MartinHugh, 2019-03-23:
I maidened a 1m fixed wing today with iNav 2.1.

Manual trim seemed OK.  
The board seemed very close to level in angle mode.
So I proceeded with Servo Auto Trim.  

I flew straight and level and then engages autotrim, Autotrim was shown on the OSD and then about 1-2 s after engaging, the aircraft entered a steep dive.  I left it for a couple of seconds and then disegaged auto trim rather than splat into the ground.

I retried this another two times with the same results.

Any ideas what I am missing here ?

Thanks.

wx4cb, 2019-03-23:
Leave it on and reset ur manual trims to zero. Thats what i do

giacomo892, 2019-03-23:
When you disable Auto Trim after having enabled it the I term of the PIFF controller needs to start growing again since your plane can be untrimmed. 

This can cause a bit of delay in controlling the craft expecially if P and FF components are low. This should be just a quick transient.

Do you have a blackbox log?

MartinHugh, 2019-03-23:
Unfortunately not. Not a flash enabled FC

MartinHugh, 2019-03-24:
wx4cb, not sure how I can do that. The aircraft was trimmed out already.

wx4cb, 2019-03-24:
@MartinHugh you flew it in manual mode and the trimmed with the trim post/sliders correct? then you went into auto trim. once it does its thing you need to return the radio trims to zero. at least that's been my experience

MartinHugh, 2019-03-24:
@wx4cb thanks for clarifying.  Seems I really dont understand the process :-)

If I return the trims to zero position, the aircraft is now not trimmed and will not fly straight and level without me intervening using the stick.

Is this what I should be doing in autotrim, correcting the trim by holding the stick, with the trims centred ?

wx4cb, 2019-03-24:
no once you have it trimmed and have gone through auto trim, you no longer need it in the radio as it's been set int he FC so the FC will have all the trimming. so if you leave it in the radio it's basically "over trimmed" if that makes sense... even in manual mode the FC still uses the trims that are set.

if that makes sense

MartinHugh, 2019-03-24:
@wx4cb thanks. Possibly coming at this from the wrong approach, having used Vectors in the past, where any trims need to be known by the FC - but are set on the radio.  I will try again.

shellixyz, 2019-03-24:
What I do is fly in angle mode at cruise throttle with the angle of attack (pitch alignment) set correctly on a calm day then enable servo autotrim. Then land, disarm and disable servo autotrim. The plane should be perfectly trimmed.

MartinHugh, 2019-03-24:
@shellixyz, thanks will bear that in mind.  Sounds easy :-)

shellixyz, 2019-03-24:
Also after you do the autotrim you should adjust the servo endpoints (min/max) so that you have the same throws left/right and consistent up/down too. Autotrim only adjusts the servo mids. We should do something about that.

For example if you had 1000/1500/2000 as mix/mid/max for both elevons before autotrim after autotrim it might look like that left: 1100/1550/1900, right 1100/1580/1900. You should adjust the min/max like that: left: 1150/1950, right 1180/1980 to get the same max throws as before the autotrim

MartinHugh, 2019-04-06:
@shellixyz.  I did as you suggested and flew in angle mode with the board level for normal flight.  I cant say it was a perfectly calm day, but one of those might take some time :-)

This time, there was no nose dive after applying AutoTrim.  I let it fly for 5 or 10 secs and then landed, disarmed and turned off autotrim.  It flies much nicer now.

Can you clear one thing up, as different flight controllers do things differently.  
Should it be the case that the FC holds the trim, and radio is not trimmed?  Even in manual mode?

If I need to retrim in manual mode, what is best practise to reconfigure the aircraft and the FC.

Thanks

wx4cb, 2019-04-06:
@MartinHugh  here's what I do..

trim in manual. run auto trim, clear radio trims, then if i need to do any tweaking cus of the wing then i'll use the radio trims as I fly in manual 99% of the time anyway. as far as rth or cruise, the FC will handle that by itself.

MartinHugh, 2019-04-06:
@wx4cb thanks for your comments.

shellixyz, 2019-04-06:
@MartinHugh Nice ! The resulting trim is used in all modes. The difference between manual and the others modes is the PID loop. The integrator in the PID loop is basically constantly trimming your wing. Manual mode is manual there is no PID loop involved. Your trims are probably still not perfect, that's all.

MartinHugh, 2019-04-06:
@shellixyz Thanks for your help.

MartinHugh, 2019-06-05:
So this is getting flagged as about to go stale. What do I have to do to prevent that happening and to get this feature implemented?

Thanks.

closed 2019-08-18 by stale[bot]