iNavFlight/inav #4535: inav 2.1.4 set rssi_scale invalid command?

jnb81, 2019-03-24:
Hi, 

I went from inav 1.9 to 2.1.4 and it seems everything is ok even adc rssi value... but can the value be rescaled?
When I type set rssi_scale I got invalid command message. 

Also diff all missing  set rssi_scale 
It also missing at CLI commands https://github.com/iNavFlight/inav/blob/master/docs/Cli.md
Was there any change? How the value can be configured?

Jetrell, 2019-03-24:
Use `rssi_min` and `rssi_max`.  
The values can also be inverted.  e.g. rssi_max = 0  and rssi_min = 100 
This can be useful when using EzUHF with the RSSI or LQ on a PPM channel.

jnb81, 2019-03-24:
Ok thanx, so basically rssi_scale command was replaced by
rssi_min and rssi_max commands?

jnb81, 2019-03-24:
I see that... 
https://github.com/iNavFlight/inav/pull/3613
I'm sorry for spamming 
I've just tested it and it works like a charm, thanks again.

closed 2019-03-24 by jnb81