iNavFlight/inav #4536: I can't get poshold and RTH to work on fixed wing

Shagma, 2019-03-24:
I need some help. I have uploaded a video to youtube which I hope someone can take a look at: [https://youtu.be/Wg6MbViaNd0](url)

The plane is an Easy Star 2 with 3 channels (elevator+rudder+motor). Using Matek F411 Wing.
At best, Poshold makes the plane do figure eight, but often it just goes off in random direction. When engaging RTH it also seems to have no clue about direction. Either it doesn't know which direction to go in or it doesn't know how to get to that direction or maybe both.

I go from AH to PH+AH. I'm getting mixed information about whether I should do AH+PH or just PH. This thread says to only do PH as it includes AH and in fact doing both at the same time makes PH not work: [https://github.com/iNavFlight/inav/issues/3605](url)
On the other hand, the official wiki says:

> Please note that you have to use this with ALTHOLD to get a full 3D position hold!

[https://github.com/iNavFlight/inav/wiki/Navigation-modes#nav-poshold---horizontal-position-hold](url)

Although everyone I've asked says iNav works with 3ch, I still wonder if I should try to add ailerons and see what that does to this issue.

shellixyz, 2019-03-24:
You don't need to enable AH with PH starting from 2.0. I updated the wiki.

It should work with 3 channels but I think you should use stabilized roll as input source for your rudder. That could be your issue. It should use the rudder since in navigation modes the turn assistant is automatically enabled but I can't confirm that and the rudder in any case is not the primary control surface the navigation system is playing with.

Also make sure `nav_fw_cruise_thr` and `nav_fw_pitch2thr` are high enough so that your plane doesn't stall.

Shagma, 2019-03-24:
Would it make sense to have both stab yaw and roll on servo 4?

shellixyz, 2019-03-24:
You probably shouldn't do that

Shagma, 2019-04-21:
I can confirm that using stabilised roll for rudder works with automated flight modes. I think there should be an entry in the wiki on this as I spent a lot of time trying other options. Thanks for your help!

closed 2019-05-13 by Shagma