iNavFlight/inav #4539: nav_extra_arming_safety = OFF does not work

twistedwings, 2019-03-24:
After a few seconds "hardware health" in configurator turns red and arming is prevented
until gps fix.

iNav 2.1.0

SPRacingF3

shellixyz, 2019-03-24:
Your diagnostic is obviously wrong. nav_extra_arming_safety disables any navigation arming checks including GPS fix. Check in the CLI with the status command what are the reasons you can't arm.

digitalentity, 2019-03-25:
If you have GPS configured but disconnected - it will be treated as hardware failure and `nav_extra_arming_safety` will not be considered. `nav_extra_arming_safety` assumes hardware is all connected and working.

twistedwings, 2019-03-25:
Seems I have chosen the wrong gps protocoll. NMEA instead of UBLOX works.

closed 2019-03-25 by twistedwings

giacomo892, 2019-03-25:
For your good just read a couple of times more what @digitalentity wrote instead of finding hilarious workarounds to unesitant issues

rotorrad, 2021-03-02:
Hi guys.
I completely green to this and it's my first mini I've built. Why even tho I've got everything blue and it's all connected and working when i type this command it comes up with red unknown command try hep? I've copy and pasted it. same thing