iNavFlight/inav #4540: S-port Telemetry not working on WingFC board - Inav 2.1

leocopter, 2019-03-24:
## Current S-Port telem not available on WingFC board.

## Steps to Reproduce

1. WingFC board flashed with Inav2.1
2. Using FRSKY X4RSB Rx 
3. sbus on UART 3 
4. I tried Smart-Port on UART 5 and 6 with telemetry_inverted = ON and telemetry_inverted = OFF ...
5- No telemetry. 

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: Wing FC 10 DOF from Banggood.
- INAV version string:
INAV/WINGFC 2.1.0 Feb 25 2019 / 17:06:35 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

shellixyz, 2019-03-24:
You can't directly connect S.Port from anything else than a R9MM to an UART of a F4 FC. You need a bidirectional inverter or connect it to a softserial port. telemetry_inverted won't do anything on a F4 FC's hardware UART.

[image]

You can use the 3.3V from the FC instead of the 3.3V regulator.

Please join [iNav's Telegram channel](https://t.me/INAVFlight) for support

leocopter, 2019-03-24:
Right. I forgot about that. I've been using s-port telem on CH5/6 on the
Omnibus F4 with softserial 2. Is there anyway to use softserial on the
WingFC ?

shellixyz, 2019-03-24:
Yes, please try this build with softserial1 on output 6: [inav_2.1.0_WINGFC.hex.zip](https://github.com/iNavFlight/inav/files/3000857/inav_2.1.0_WINGFC.hex.zip)

shellixyz, 2019-03-24:
Yes, please try this build with softserial1 on output 6:  [inav_2.1.0_WINGFC.hex.zip](https://github.com/iNavFlight/inav/files/3000861/inav_2.1.0_WINGFC.hex.zip)

shellixyz, 2019-03-24:
Probably better to use this one I made a PR for having SS1 on UART6 then you won't have to rewire your wing on the next release if it is merged: [inav_2.1.0_WINGFC.hex.zip](https://github.com/iNavFlight/inav/files/3000872/inav_2.1.0_WINGFC.hex.zip)

leocopter, 2019-03-24:
Wow ! Thanks. I will.
I did a hack on my X4R : I take the s-port signal before the inverter.
The vendors ship this board with FF_F35_Lightning Inav 1.91 firmware. They
claim UART 6 Tx has a hardware inverter ( it is the SPO pin on the board)
and can be connected directly to S-Port.
I does not work.
I will test your custom firmware.

Regards,

Chris

leocopter, 2019-03-24:
Hi again. No luck with the custom firmware. S-Port telem works on UART 5
with my hacked (non inverted) X4R. I get no Telem on Softserial 1 (UART 6
Tx : SPO pin) with the normal S-Port.
May be there is something wrong with the routing to/UART6 port itself ?

Chris

shellixyz, 2019-03-24:
Did you set `telemetry_inverted` to ON ? It is needed for it to work on SS if you hacked the RX.

leocopter, 2019-03-24:
Yes I did. No luck so far.

leocopter, 2019-03-24:
I hacked the Rx by adding a wire to the non inverted signal. I kept the
regular connector. I can use the inverted signal and the non inverted
signal. Neither work on Softserial 1.

shellixyz, 2019-03-24:
My mistake I forgot to define timers for the SS pins. SS on UART6 will not work. Please try this build with SS1 on UART2: [inav_2.1.0_WINGFC.hex.zip](https://github.com/iNavFlight/inav/files/3000986/inav_2.1.0_WINGFC.hex.zip)

leocopter, 2019-03-24:
I will try it. A lot of people use UART 2 for GPS. Is it possible to route
ss1 to UART 4 ? Or 5 ?

Thx

Chris

shellixyz, 2019-03-25:
It is not possible the UART4/5 pins don't have timers

leocopter, 2019-03-25:
Hi Michel. S.Port telem works on  Sotfserial 1 routed to UART 2 Tx.
Thank you !
Do you think we could have LED_STRIP on another port ? ;-) (UART 4 tx or
"Air" ?)

Good night.

Chris

shellixyz, 2019-03-25:
Nice. I can't say for sure for the LED_STRIP feature. It should be possible. I don't know enough about DMA stuff to do it myself for the moment though. Please open a new feature request if you'd like it to be added.

closed 2019-03-25 by shellixyz

UnchartedBull, 2020-03-20:
Was there a follow-up on this? Would love to have LED_STRIP on the WingFC. I‘ll be using a GPS / FrSky XSR with S.Port Telemetry and SmartAudio. So something like AIR or one of either RX1/TX1/RX5/TX5 (would connect GPS to the opposite one then) would be perfect. If none of you opened an issue for that, I‘ll go ahead and open one.