iNavFlight/inav #4542: Rangefinder failing after upgrading to 2.1.0

georgekucher, 2019-03-24:
## Current Behavior

After I upgraded to INAV v2.1 my sonar don't initialize.

Sonar: US-100 (via I2C and Arduino)

# status
System Uptime: 14 seconds
Current Time: 2041-06-28T04:04:00.000+03:00
Voltage: 14.81V (4S battery - OK)
CPU Clock=216MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=UNAVAILABLE, OPFLOW=NONE, GPS=OK
SD card: Manufacturer 0x3, 15558144kB, 01/2017, v8.0, 'SL16G'
Filesystem: Ready
Stack size: 6144, Stack address: 0x20010000, Heap available: 1456
I2C Errors: 45, config size: 4671, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 5, cycle time: 2010, PID rate: 497, RX rate: 129, System rate: 9
Arming disabled flags: NAV HWFAIL CLI

## Steps to Reproduce

1. copy diff on v2.0
2. upgrade to v2.1
3. paste diff
4. save

## Expected behavior

Sonar should work.
## Suggested solution(s)

Check the code

## Additional context

https://pastebin.com/zrCncp68

---

- FC Board name and vendor: Matek F722
- INAV version string: INAV/MATEKF722 2.1.0 Feb 25 2019 / 16:59:54 (65b0ec142)

shellixyz, 2019-03-24:
I don't know why but RANGEFINDER_HCSR04_I2C has been disabled for the MATEKF722 target in 82d3dbe23. @digitalentity ?

It should work again after you install this build: [inav_2.1.0_MATEKF722.hex.zip](https://github.com/iNavFlight/inav/files/3000904/inav_2.1.0_MATEKF722.hex.zip)

georgekucher, 2019-03-24:
Thank you @shellixyz. I will try it.
Besides it is interesting to know why the sonar was disabled, and will it be enabled in 2.2

shellixyz, 2019-03-24:
I think it was disabled by mistake I will make a PR to enable it again in the next release

georgekucher, 2019-03-24:
Great. Thanks a lot.

shellixyz, 2019-04-27:
Fixed, closing

PR #4545 (2019-03-26): Enable RANGEFINDER_HCSR04_I2C for all targets with flash > 256k
Fixes #4542
closed 2019-04-27 by shellixyz