iNavFlight/inav #4543: Issue with inav furious f-35 board. Servos on volantex ranger ex servos not responding correctly.

Raptorframes, 2019-03-24:
Servos move by don’t follow the stick they all seem so twitch into place or lag behind a lot. Can show video of issue

Raptorframes, 2019-03-24:
I‘ve just uploaded a movie to YouTube!
You can check it out here:
http://www.youtube.com/watch?v=ylpw872oDik

shellixyz, 2019-03-24:
Do you have a screen connected to your FC ? Please post config dump

Raptorframes, 2019-03-24:
Here is my Dump

$X>jïÿ' Entering CLI Mode, type 'exit' to return, or 'help' # Dump

# version
# INAV/FF_F35_LIGHTNING 2.0.1 Oct 31 2018 / 18:53:37 (a04e68aaa)

# resources

# mixer

mmix reset

mmix 0 1.000 0.000 0.000 0.000
mmix 1 1.000 0.000 0.000 0.000

# servo mix
smix reset

smix 0 3 0 100 0
smix 1 4 0 100 0
smix 2 3 14 100 0
smix 3 4 14 -100 0
smix 4 5 2 100 0
smix 5 2 1 100 0

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE
feature OSD
[... 748 lines cut ...]
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

shellixyz, 2019-03-24:
I can't see anything special in there. Make sure the lag doesn't come from your RC link. There is very little distance between your TX and RX sometimes it can do that.

Raptorframes, 2019-03-24:
I can guarantee it is not that. I’m using a crossfire and have lots of experience using this transmitter/receiver. I just can’t figure out the issue!

sniperxnl, 2019-03-25:
Does this also happend in Horizon mode, when you move the plane by hand?

Raptorframes, 2019-03-26:
Yes, when I’m horrizon the servos respond the same way.

sniperxnl, 2019-03-27:
I think the problem is your type of servo! I advise you to test a diffrent brand servo.

closed 2019-05-09 by DzikuVx