iNavFlight/inav #4546: F35 Nav launch motor auto rotation!

letruong6006, 2019-03-25:
Please help me, the first step I have enabled NAV Launch mode, the next step is to turn on the arm, set the throttle value to about 50% but the motor is rotating immediately while the plane is on the face. land

sniperxnl, 2019-03-25:
if you setup inav launch on always then use my cli dumP
nav_fw_launch_velocity = 300
Allowed range: 100 - 10000

nav_fw_launch_accel = 1863
Allowed range: 1000 - 20000

nav_fw_launch_max_angle = 40
Allowed range: 5 - 180

nav_fw_launch_detect_time = 40
Allowed range: 10 - 1000

nav_fw_launch_thr = 1950
Allowed range: 1000 - 2000

nav_fw_launch_idle_thr = 1200
Allowed range: 1000 - 2000

nav_fw_launch_motor_delay = 500
Allowed range: 0 - 5000

nav_fw_launch_spinup_time = 100
Allowed range: 0 - 1000

nav_fw_launch_min_time = 0
Allowed range: 0 - 60000

nav_fw_launch_timeout = 15000
Allowed range: 0 - 60000

nav_fw_launch_max_altitude = 0
Allowed range: 0 - 60000

nav_fw_launch_climb_angle = 35
Allowed range: 5 - 45
[image]

letruong6006, 2019-03-25:
ohh. Thanks for the enthusiasm for help, I will try again, wish you a good
day

closed 2019-06-07 by stale[bot]