iNavFlight/inav #4547: Can't connect OMNIBUSF4PRO on Mac

flybabo, 2019-03-25:
I successfully programmed an OMNIBUSF4PRO with inav2.1 firmware downloaded from online using 2.1.4 inav-configurator on a MacBook Pro.
But I can't connect to the FC after power cycling.
inav-configurator complains that "No configuration received within 10 seconds, communication failed"  I've never have had such a problem before.
I tried the Betaflight firmware in the same manner but it works fine with this FC on the same MacBook Pro.
Any clues?

malckris, 2019-03-25:
Check the port selection at the top of the screen... same issue for me, Betaflight always connects but iNav on my mac reverts to bluetooth after a flash and I need to select the USB connection, after that it remembers the connection....

flybabo, 2019-03-25:
inav-configurator automatically detects the usb connection when I plugged-in an OMNIBUSF4PRO just like the betaflight configurator.
I even selected usb connection manually, but result stays the same - 10 seconds timeout.

Troynica, 2019-03-26:
This happened to me when I enabled OSD when no OSD hardware was present. It may be the same for other (SPI based?) hardware that is configured but not present/working.

shellixyz, 2019-03-30:
Most likely an issue with the configurator or drivers on iOS. I'm closing this issue and leaving the one you opened in the configurator project open.

closed 2019-03-30 by shellixyz