iNavFlight/inav #4548: Can't get Position hold and RTH work consistantly

Ralfde, 2019-03-26:
Airplan with V-Tail with Omnibus F4V3
[inav.zip](https://github.com/iNavFlight/inav/files/3007556/inav.zip)

The logfile does show the modes Position hold and RTH correctly, but in the OSD display it's not shown and also the rudder behaviour does not show this functions working at all. They just still work in level mode.

A few times it worked randomly and then it was also shown in the OSD display. At the moment I have no idea any more, why this modes do not work and why there is no error message in the OSD, what could be maybe wrong.

shellixyz, 2019-03-26:
First not related to your issue but your settings look a bit weird to me:

```
set acc_lpf_hz = 5 # that's very low
set nav_fw_cruise_thr = 1750 # that's pretty high but why not I guess
set nav_fw_pitch2thr = 0 # disabled pitch to throttle compensation, why ?
```

also
```
set nav_fw_pos_xy_p = 200 # very high the default is 75
set nav_fw_pos_xy_i = 35 # very high too the default is 5
set nav_fw_pos_xy_d = 10 # ok I guess the default is 8
```

Please start with the default values for these settings they should work fine for most planes.

That's not right. Capacity thresholds are remaining capacity. 840 for warning seems pretty high. The critical threshold should be bellow the warning threshold.
```
set battery_capacity = 1200
set battery_capacity_warning = 840
set battery_capacity_critical = 1080
```

You don't need a compass on planes. More than that it is not recommended.

The rudder is only used for turn coordination.

The log shows that your plane was on the ground. You can't see if it is working or not like that

Ralfde, 2019-03-26:
Sorry, I didn't have the SD-card installed during the flight, but I had the same problem during the flight and the airplane did never come home. So I wanted to check at least the correct mode switching on the ground and also the rudder movement can be check at least if it works principell. 
I wired up now the compass, too and now the mode switching seem to work correct every time.
Does the mode switching without compass maybe need a minimum GPS speed or minimum altitude?
I will test next, if coming home and position hold works now with compass. In case it works correct I will deactivate the compass and try it without again also.

Capacity thresholds could be one of the problems. So the warning should be at f.e. 20% and the critical capacity f.e. at 10%?

Ralfde, 2019-03-27:
I changed now back to the standard values and I corrected the capacity thresholds. But still the same problems. inav switches correctly into the flight phases Pos hold and RTH, if the compass is activ. When the compass is not activ, it does switch into AH (Altitude Hold), but not into position hold or RTH, but the logfile does show it correct. The OSD not and also the rudder reaction is only like Altitude hold.

Ralfde, 2019-03-27:
The Airplane does want to crash just direction ground in position hold and RTH mode. I have no idea why.
I hope somebody can help why. Here the log file:
[LOG00012.zip](https://github.com/iNavFlight/inav/files/3013474/LOG00012.zip)

stronnag, 2019-03-27:
If you're going to use a compass, at least make sure it works correctly; this one appears to not to. The heading and CoG sufficiently to trigger mwp's naive heading anomaly detector repeatedly. Replay visualisation: https://youtu.be/dqyeCmEbeH8

Ralfde, 2019-03-27:
IST8310 direction looked good at the ground and at the start. Maybe to much influence of the motor current.
So I have no other chance to try without compass again and make another log.
Would be nice, if the heading error message would be also shown in the OSD.

Ralfde, 2019-03-28:
[inav3.zip](https://github.com/iNavFlight/inav/files/3018241/inav3.zip)

I tested now inav 2.01 and 2.10 without compass. But the problem is about the same.
The airplane is sinking fastly and does not make direction corrections. The only difference today, that at some tries, at the beginning the function seem to work and after latest half a circle, the airplane did go again straight direction to the ground.
I think there meight be a hidden bug, because sometimes for short time the function seem to work, before it does strange things and only going back to the angle mode does protect the airplane before crashing.

Ralfde, 2019-04-01:
[inav_F7NXt.zip](https://github.com/iNavFlight/inav/files/3029533/inav_F7NXt.zip)

Tested now the OmnibusF7NXT. Same problem. Sometimes position hold seem to work for a while, but after some time the Airplane does sink with the max. allowed pitch angle. I have no idea, why. Sometimes it does sink from the beginning.

digitalentity, 2019-04-01:
My blind guess is configuration issue or hardware problem.
Will have a look at the logs and share my findings here.

digitalentity, 2019-04-01:
@Ralfde in the last log which part of the flight is the full-pitch dive?

I noticed an interesting thing there during RTH phase. Even with full pull-up (20 deg) plane is barely able to keep altitude and definitely unable to climb to requested 70m. Is this actual flying or you are testing on the bench?

digitalentity, 2019-04-01:
I also suspect we have a case of misaligned flight controller together with incorrect stick or servo inversion.

RTH requests climb (pitch stick down), plane seems to be reaching the climb angle correctly (from FC point of view), however altitude is declining rapidly as if the plane is in fact diving.

This may happen when flight controller is mounted 180 deg incorrectly and the problem is "mitigated" with servo and stick inversion in the radio. 

When flying in ANGLE mode this may seem to work fine:
- Pilot PULLS the pitch stick
- FC treats this as DIVE command due to stick inversion and happily executes
- Due to servo inversion and FC misalignment this causes the plane to actually CLIMB, however from FC perspective plane in DIVING as commanded.

This will totally break the RTH though, because FC will get a CLIMB command, but due to servo inversion will be actually DIVING without knowing it. It will loose altitude and will try to CLIMB even harder and eventually will reach full dive.

There is a nice video with explanation of common INAV setup mistakes by @Painless360: https://youtu.be/fVJYodLimD8

Troubleshooting steps: 
1. Verify that when you push sticks to up-right position you read 2000 in 4 first channel in Receiver tab
2. Verify that model in the Configurator reflects the motion of the plane
3. Once 1 and 2 are verified to be correct - make sure servos respond correctly in MANUAL, ACRO and ANGLE modes.

Only then you should try flying.

Ralfde, 2019-04-02:
Yes, the elevator direction in the radio was reversed and also the build in direction of the FC was wrong.
I suggest to include this video link to the inav configurator. I believe this will help in a lot of cases.
Looking forward to the next test flight.
Thanks a lot for your help.

Ralfde, 2019-04-02:
Flight tests worked great. Thanks a lot for the support.

closed 2019-04-02 by Ralfde

digitalentity, 2019-04-02:
No worries, thanks for using INAV 😄