iNavFlight/inav #4550: Logic conditions

hali9, 2019-03-27:
A some idea from https://github.com/iNavFlight/inav/pull/4533
1. Input logic one
2. BETWEEN operator
3. Permit in smix to put more that one condition using & operator

DzikuVx, 2019-03-27:
On the topic of `3` -> Allow LCs to accept other LCs. At the end `smix` will be always activated by one LC

DzikuVx, 2019-03-27:
And `ONE` should have internal value of `500`.

hali9, 2019-03-27:
https://github.com/iNavFlight/inav/compare/development...hali9:logic-condition-one

hali9, 2019-03-28:
@DzikuVx If You want take a look
https://github.com/iNavFlight/inav/compare/development...hali9:logic-condition-1

DzikuVx, 2019-03-28:
@hali9 I do not think it looks good with operand C. It makes everything so complex and consumes even more RAM than before. My idea was rather the following:

If a LC can accept a different LC as an operand the following is possible:
* `LC 0` defines RX channel is > 1400
* `LC 1` defines RX channel is < 1600
* `LC 2` defines that `LC 0` and `LC 1` are true
* `smix` is activated when `LC 2` is true

This way LCs are still 2 operand entities like they should be and all more complex conditions can be built using LCs as inputs to diffent LCs. 
I have most of the GUI for LC ready and changing it to use 3 operand would be a major PITA. 
Let me propose a PR that accepts LCs as arguments. How about that?

DzikuVx, 2019-03-28:
@hali9 look at #4555

hali9, 2019-03-28:
Quick attention: I have 3 LC (using beetween). Now instead 1 LC I have 3 LC, so I need 9 LC.
The logic list is limited to 8, this is realy low. 
This will ok as between operator if logic list count will be high 16 for example.
>     * `LC 2` defines that `LC 0` and `LC 1` are true
how we will compare it: true > true, true == false ?
do we need creaate operators `and` `or` (`xor` `not`) in  logicOperation ?

DzikuVx, 2019-03-28:
we can bump the number of logic conditions without problem. Only F3 suffer from low RAM and it's not enabled on F3 either way. So yes, we can bump it to 16 without any problems

DzikuVx, 2019-03-28:
`true > true` is false
`true == false` is false

now that you mention it, AND and OR are missing. They can be easily added.

Let me make a PR for that too

hali9, 2019-03-28:
So true == false con wor like xor

DzikuVx, 2019-03-28:
Looks like. But indeed AND OR NOT and XOR operators are a way to go

hali9, 2019-03-30:
@DzikuVx Thank for Your cooperation.

closed 2019-03-30 by hali9

DzikuVx, 2019-03-30:
@hali9 my pleasure