iNavFlight/inav #4553: GPS not working on a wireless connection to iNav

Tyson-D, 2019-03-27:
If the title was rather confusing, let me elaborate a little bit on what my issue is:

I'm building a flying wing RC plane with a Matek F405 Wing flight controller, a Holybro Micro M8N GPS Module, and a pair of standard 915MHz 3DR telemetry radios. I am able to connect my flight controller to the iNav Configurator V2.1.4 both with a USB cable and with my telemetry radios. The radios are using SiK firmware 2.0 modded with MSP capabilities. 

As for the GPS, I am able to get and maintain a steady 3D fix and flow of data when the flight controller is connected to my computer via USB. However, when I try to get a fix on my GPS on a wireless connection (through the telemetry modules), I am receiving no messages nor signs of communication from my GPS, as shown in the linked image. Even when I don't have a fix, the "Total Messages" counter on the GPS tab in the configurator should be ticking up as soon as the radios are connected, but the number remains at zero. 

The hardware indicators for both the GPS and Compass are both blue on the iNav Configurator, but I am getting no data from either of them. I have tried using various UART ports for both the GPS and the telemetry modules, but to no avail. I also made sure that I was not using any UART that interfered with the I2C port I am using for my compass (number 2). Currently I am using UART 3 for my telemetry connection and UART 5 for GPS.

Any help for solving my issue would be much appreciated, because quite frankly, I have run out of ideas. Sorry for the wall of text.

[image]

[image]

wx4cb, 2019-03-27:
following this one, i'd be interested to know as well (i'm doing a similar thing but using crossfire). I've managed to get it to work using mavlink and mission planner but not with MSP. when it's not raining here, i'll give it a try the way you have it setup.

shellixyz, 2019-03-28:
I don't know how the "SiK firmware 2.0 modded with MSP capabilities" is working but maybe it doesn't support the MSP message used to fetch the GPS coordinates. `MSP_RAW_GPS` is used for that.

stronnag, 2019-03-28:
The sik firmware variant merely optimises 3dr framing for mspv1. it does not care about individual message types. With the advent of mspv2 it provides no benefit.

For the record, inav / MSP over 3dr works perfectly for me.

Tyson-D, 2019-03-28:
stronnag, could you maybe tell me the steps you took to install your gps and telemetry radios? Maybe I'll find a discrepancy between my installation and yours.

stronnag, 2019-03-28:
serial on a "WingFC" using 3DR for telemetry

```
# uart 1 is the 3DR (MSP unarmed, LTM armed)
serial 0 17 115200 115200 115200 115200
# uart 2 is the BN880 GPS
serial 1 2 115200 115200 0 115200
# uart3 is the RX
serial 2 64 115200 115200 0 115200
serial 3 0 115200 115200 115200 115200
serial 4 0 115200 115200 115200 115200
serial 5 0 115200 115200 0 115200
```
3DR settings:
```
S0:FORMAT=26
S1:SERIAL_SPEED=115
S2:AIR_SPEED=64
S3:NETID=197
S4:TXPOWER=20
S5:ECC=0
S6:MAVLINK=0
S7:OPPRESEND=0
S8:MIN_FREQ=433050
S9:MAX_FREQ=434790
S10:NUM_CHANNELS=20
S11:DUTY_CYCLE=100
S12:LBT_RSSI=0
S13:MANCHESTER=0
S14:RTSCTS=0
S15:MAX_WINDOW=66
```
Using the 'standard' firmware. I ceased maintaining the MSP variant once we introduced MSPv2. But using the SiK/MSP variant is harmless, it works too.

wx4cb, 2019-03-28:
@stronnag any reason to use ltm over mavlink? what ground software are you running?

Tyson-D, 2019-03-29:
@stronnag I think the only notable differences that I found were on your radios -- My MAVLINK parameter is turned on and you're using a baud rate of 112, whereas mine is 57. I don't think either of these should make a difference on what is causing my issue, but I do have a question: How did you change the 3DR message protocol from MSP when unarmed to LTM when armed? I didn't know you could do that. 

As an update, I have been playing around with the CLI to try and find any solutions in there and I noticed that when I use the command "gpspassthrough" it seems like I am getting the same message/string of characters about once per second. I'm curious to know if the GPS is sending actual messages back to iNav and not showing up on the GPS tab in the [configurator.]

[image]

stronnag, 2019-03-29:
> @stronnag any reason to use ltm over mavlink? what ground software are you running?

Bandwidth, throughput, better support for iNav navigation modes.
Bizarrely, I use mwp https://github.com/stronnag/mwptools for GCS.

stronnag, 2019-03-29:
> @stronnag I think the only notable differences that I found were on your radios -- My MAVLINK parameter is turned on and you're using a baud rate of 112, whereas mine is 57. I don't think either of [...]

If you set MSP on a UART and a telemetry protocol (LTM, MAVLINK etc) and enable `feature TELEMETRY`, then you get MSP unarmed and the telemetry protocol when armed. You need a ground station that can automatically detect and switch protocols as well (ezgui, mission planner for iNav, mwp all can do this, the configurator does not); and the same baud rate for both MSP and telemetry.

> As an update, I have been playing around with the CLI to try and find any solutions in there and I noticed that when I use the command "gpspassthrough" it seems like I am getting the same [...]

That is the documented behaviour of `gpspassthrough`; it allows you to "connect" the GPS to a analytic tool like `u-center` or `ublox-cli` for testing / debugging. It can only be engaged in the CLI and requires rebooting the FC to get back to normal mode.

Tyson-D, 2019-04-09:
Update: I have achieved a 3D fix on iNav through telemetry. The problem was that on the "Ports" page in the configurator, I had not enabled "MAVLink" under the telemetry column. My guess is that the flight controller didn't have the instructions to carry the GPS information to the telemetry module. So, according to the Setup page, I was able to get and maintain a 3D fix with up to 12 satellites, to my delight. 

However, I can't seem to stop running into problems: On the Setup page, under the GPS section, the configurator clearly states that the GPS module has a 3D fix and updates its longitude and latitude. However, on the GPS page, I have the same screen with all 0's instead of any actual data. This seems to be a problem with the configurator, as when I do the same with a wired connection to the FC I can easily find my location on the map in the GPS page, as well as all of the data I need. Thank you all for helping me two weeks ago, and if anyone cares enough to revisit this thread, that would be great.

wx4cb, 2019-04-09:
if it's a configurator issue, you need to post an "issue" in the configurator repo.

closed 2019-04-10 by Tyson-D

wdliu356, 2023-04-13:
I know this is a super old thread but just wanna check if you succeeded in getting the GPS shown on the GPS page in the end