iNavFlight/inav #4554: RTH backward orientation

Gabbocombinaguai, 2019-03-27:
Hello everyone.

My issue is the following.

When I switch to RTH mode the drone goes backward and the compass is 180° turned.

What i did before:
the gps far from wires (20cm far)
I double cecked every wire of gps
Gps orientation is pointing the right direction( as the manual said). the model is  bn 880
Magnetometer calibration has been done correctly
the Magnetometer Declination, Done.

Once I joined the inav setup i tryed every option possible 0°, 90 CW, 180° CW, 270° CW, 90° CCW, 180CCW, 270 CCW.
The right one is 90° CCW, because it is the only one that when i switch to pos and alt hold the drone is holding the right place ( i mean no movement like crazy)

The RTH is working fine( 90° CCW), when I switch that, the drone is coming back to arming position and start descending, but it comes backward and the compass is 180° turned.

Any suggestion?

closed 2019-03-27 by Gabbocombinaguai

teckel12, 2019-03-28:
@Gabbocombinaguai Can I guess? 270 Flip or 90 Flip?

Gabbocombinaguai, 2019-03-28:
My fault. I understood the mistake. I wrongly installed the fc with the arrow pointing towards the tail :)
Thanks anyways as usual.