iNavFlight/inav #4559: FrSky X4RSB Receiver just not working in iNav with my Matek 405-Wing FC

shresthanator, 2019-03-28:
Issue: I'm an experienced RC (plane) hobbyist, but brand new to iNav and flight controllers in general. I just cannot get my RX to work with my FC in iNav. 

Hardware

* [Matek F405-Wing Flight Controller](https://www.getfpv.com/matek-f405-wing-flight-controller.html)
* [FrSky X4RSB 3/16CH Telemetry Receiver](https://www.amazon.com/gp/product/B00VM1XWB0/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)

Before I plugged in the RX to my FC, I verified that my RX is properly working and bound by testing it out on a servo. Everything seems to be in order. I hooked up the RX to my FC following that wiring guide in the first link listed above - ground, 5v, Sbus, and smartport wires all connected to the proper pins (hard to mess this part up).   

Note the RX powers on and the green-light indicates that the RX is properly bound to my transmitter. 

I've scoured [online resources](https://www.youtube.com/watch?v=-cu8gO9px8M) to debug my issue, but no matter what I have done I cannot seem to resolve my issue. When I move around the control sticks, iNav gives no indication that there is anything happening in the "Reciver" tab- the channel indicators do not move at all.   

 I have proper iNav Configuration Settings:

* Ports Tab: 
  UART2 is set to "Serial RX"
  SOFTSERIAL1 is set to "SmartPort"

* Configuration Tab: 
   Verified that "Enable CPU based serial ports" and "Telemetry Output" are both on. 
   "Serial based receiver" is selected and "SBUS" is selected as the serial based provider

It is very possible that I am overlooking something very simple. Can someone please help me out?

DzikuVx, 2019-03-28:
CLI dump please

shresthanator, 2019-03-28:
> 
> 
> CLI dump please

# profile 0

# dump

# version
# INAV/MATEKF405SE 2.1.0 Feb 10 2019 / 16:44:59 (0d3d0ff30)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature SOFTSERIAL
feature GPS
[... 936 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

shresthanator, 2019-03-28:
> 
> 
> CLI dump please

EDIT: I was able to get the receiver working - the receiver tab is now responsive. However, the channel mapping is mixed up: 

TX Throttle = Roll on iNav
TX Yaw = Yaw on iNav (correct)
TX Pitch = Throttle on iNav
TX Roll = Pitch on iNav

Can you please help me out? 

here is the cli dump:

# dump

# version
# INAV/MATEKF405SE 2.1.0 Feb 25 2019 / 17:02:18 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature SOFTSERIAL
[... 937 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

wx4cb, 2019-03-28:
in the receiver tab change the channel map to TAER1234 or whatever matches
your transmitter

DzikuVx, 2019-03-29:
Or even better in the radio change the channel maping to AETR which is the default value

closed 2019-03-29 by DzikuVx