iNavFlight/inav #4560: CRSF LQ scaling on CH17 is somewhat broken

tgreer, 2019-03-29:
## Current Behaviour
The current code seems to scale LQ (which is 300 to 0) to 0-100. This doesn't seem to be right. It's not quite a linear scale either. What's displayed is extremely confusing for pilots who understand CRSF LQ.

## Steps to Reproduce
use CRSF. Display RSSI. Recieve nonsense values.

## Expected behaviour
LQ should be taken as a raw figure from the telemetry and displayed as an OSD item instead of messing about converting it to 1000-2000

## Suggested solution(s)
Take raw value from linkStats->uplinkLQ and use that for new LQ variable on OSD.

## Additional context
https://github.com/iNavFlight/inav/blob/65b0ec142c5b27bb5f0e8ea14f8d0627cff7ca76/src/main/rx/crsf.c#L234

---
- FC Board name and vendor:
Omnibus F4 Pro V3
- INAV version string:
# version
# INAV/OMNIBUSF4PRO 2.1.0 Feb 25 2019 / 16:53:27 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

teckel12, 2019-03-29:
@tgreer I believe Crossfire sends the value back to the FC in the 0-100 range.  I know that LQ for Crossfire is some bizarre 0-300 range, but what Crossfire actually sends back is 0-100 (because the FC and every other protocol is expecting a 0-100 range).

The code you snipped just scales the 0-100 to the 1000-2000 range of a channel.

tgreer, 2019-03-29:
> @tgreer I believe Crossfire sends the value back to the FC in the 0-100 range. I know that LQ for Crossfire is some bizarre 0-300 range, but what Crossfire actually sends back is 0-100 (because the [...]

Well somethings not right somewhere. The output on say CH12 if i map LQ there and the output on CH17 doesn't match. I believe it's sending 0-300 in the frame and that code is scaling it to 0-100

teckel12, 2019-03-29:
@tgreer My Crossfire LQ is mapping correctly.  I'm sending LQ back over channel 12 and I'm showing a 0-100 RSSI scale in the configurator and OSD.

tgreer, 2019-03-29:
Correct, but LQ is now also sent over CH17, and its wrong.

teckel12, 2019-03-29:
@tgreer What do you mean by output on CH17?

tgreer, 2019-03-29:
If you read the code I linked or double check, LQ is pulled direct from the CRSF frame and fed onto CH17 by inav.

Regards

Thomas

teckel12, 2019-03-29:
@tgreer So this is to do with the MSP LQ that's mapping to a fake channel 17.  Is this new because I always used channel 12 from the transmitter module.  Anyway, that was my confusion.  

I'll have to try this to see what the range is.  If it's a 0-300 range it's a simple enough fix (change the `100` to `300`).  I still probably would use the LQ sent back on channel 12, because the Crossfire 0-300 range is total crap (it sticks at 300 most of the time, rendering the data worthless).  I've been told by TBS to use the RQ/RSSI sent over channel 12 as that's more accurate than the 0-300 LQ nonsense (which I believe is mostly marking hype).

wx4cb, 2019-03-29:
it can't be coming from crossfire as there's only 12 channels. LQ on 12 works fine, how do you get it onto ch 17 and i can check

tgreer, 2019-03-29:
> it can't be coming from crossfire as there's only 12 channels. LQ on 12 works fine, how do you get it onto ch 17 and i can check

Literally use CRSF, see channel 17 in the Reciever/Radio tab.

MartinHugh, 2019-03-29:
Pretty sure that it was TBSs intention to allow this value to be "summarised" in a 0-100% scale for OpenTx use.  Not sure if that is relevant to either of the channels you are looking at:

From Crossfire RX version history

2.31 2018-04-21
- added: LQ is shared in new format for OpenTX telemetry and receiver output. Recommended
warning level is around 70%. RFMD can be excluded for OpenTX remotes warnings

wx4cb, 2019-03-29:
@tgreer interesting.. I didn't know that... i'll take a look when i go flying out tomorrow

wx4cb, 2019-03-29:
@MartinHugh i beleive that has since changed in the latest releases (2.93 is the latest i beleive). From something I saw in the TBS Facebook group. @teckel12  might be able to shed more light on it.

MartinHugh, 2019-03-29:
2.94 is the latest at the moment I believe.  I could be missing something, but I cant see anything relating to LQ in the version history for 2.93 or 2.94 (or anything in fact since 2.31).  I fly on 2.40 and it is a percentage and there is no longer any need to alarm when RFMD changes (although I still find it useful from a signal degradtion point of view to have the tx announce it when it goes from 2 to 1 and back)

teckel12, 2019-03-29:
Crossfire is only 12 channels. This 17th channel is faked. The LQ is sent back to the FC via MSP. This didn't function till I added MSP over crossfire in the latest INAV release. Unknown to me, the LQ is faked as channel 17. But, it seems it's jacked up (probably because it could never be tested).

I can look at it in a few days and it shouldn't be much to fix. But, I'd suggest not using it because LQ is kinda bogus IMHO. When in there poking around, I may see about adding a link quality value that has more value. Maybe adding that to another channel or a switch to use either LQ or a more linear value that tells the pilot something.

Jetrell, 2019-03-29:
@teckel12  Are you sure about CH17 being fake?
It is there. But L3 has to be enabled and calibrated for it to show up. 
Its the channel that the rotary switch is designated too.. If one chooses to install the hardware.

teckel12, 2019-03-29:
@Jetrell It's injected in channel 17 on the FC with INAV. It doesn't send on a channel because there's no channel 17 on Crossfire. It takes the LQ message and creates a channel 17 which appears exactly like a real channel.

FPVZaphod, 2019-05-07:
Wouldn't it be better to make it possible to choose the source of ch17 in the Configurator? 
To make it users choice if ch17 shows RSSI, LQ or LQ/RSSI? Or even one of the other link related informations provided by the crossfire?

teckel12, 2019-05-08:
@FPVZaphod I think people are too hug up on the importance of RSSI vs LQ with Crossfire.

Anyway... Can't you select if you want RSSI, sq, or both?

FPVZaphod, 2019-05-08:
Yes, but only when I use a normal Crossfire Channel, and then I usually switch it to RSSI.
This magic channel 17 is a great solution, just LQ is maybe not the best choice. But I will try, maybe I just need to get used to it :-)

MartinHugh, 2019-05-08:
@teckel12 I think it depends on the environment in which you fly.  LQ is much more useful than RSSI. The noise floor where I fly varies a lot depending on whether you happen to be in line with a TETRA tower 2km away. Flying with LQ instead of RSSI gives me a meaningful reading about the control link wherever I go - unlike RSSI.  But I agree that for low noise floor environments the difference in usefulness between them is small.

teckel12, 2019-05-08:
@MartinHugh I just typically get all 100% LQ and then it drops really low. Basically, like little warning or not very linear at least.  The RSSI on Crossfire is kinda useless as it's invented and a 130dB scale, and when I've tried using it I get nothing but complaints as everyone flying CRSF expects the signal quality to be 100% all the time. 

With RSSI using FrSky, I find it much more realistic and linear.

closed 2019-09-19 by stale[bot]

Pairan, 2020-07-13:
I would like to reanimate the topic-  Just read all of it!

If im right then the problem still exists - putting LQ on channel 12 in CF won't output the same as (fake) channel 17 does while using crossfire.

It would be kinda good if the value shows the same. Also I would like the suggestion made above:

> Wouldn't it be better to make it possible to choose the source of ch17 in the Configurator?
>To make it users choice if ch17 shows RSSI, LQ or LQ/RSSI? Or even one of the other link related informations provided by the crossfire?

## Why did I end up here and why would a change be good?
Well, Crossfire only has 12 channels. And I will have to scarify a feature no my quad because I want the stuff LQ would output and map to channel 12. If this value on the other hand would be the same as in the fake channel 17, then I would be able to control my RunCam Hybrid ...

wx4cb, 2020-07-13:
As stated multiple times by trappy.... using rssi is pretty much irrelevant. Also,i beleive that you just need to use the LQ osd item and it will show correctly. I am not sure that using ch17 is even a thing any more. I dont think ive even got it setup on mine.  Not bear any models for a couple days becore i could check

wx4cb, 2020-07-13:
Heres something that was posted by mike Chin recently on facebook

this might help a few people 
“Why LQ on Crossfire and what does it mean?”
Crossfire is a much different system than the Traditional RC systems you have likely previously used in RC. It offers a lot more information for you to be able to Pilot your craft and know what’s going on, especially with how strong your link is.
LQ = Link Quality, this is the parameter that matters on crossfire to tell how strong your connection to your receiver is
RSSI= Raw Signal Strength Indicator- while this is a useful parameter for someone who is experienced, for MOST pilots this reading will only confuse you, especially if you are coming from Frsky who uses RSSI as their indicator for link.
When do I turn around using LQ? = 70%
A good analogy for this is to think of the Crossfire Receiver  as a pair of `ears, and the Transmitter as a loudspeaker. The TBS crossfire receiver has super sensitive superpower ears. At range the loud speaker may have a low volume(RSSI) but the receiver can still hear the loudspeaker and understand everything its saying(LQ)
Why do I use LQ instead of RSSI on crossfire?
To know the range limit based on RSSI the noise floor needs to be known as well. Noise floor depends on the environment and on the components built-in the airframe. If you are not so familiar with how RSSI works we recommend to use LQ.
It seems most people coming from frsky do not understand how rssi works. its not a linear #, if you're looking for a good indication of signal LQ is what you should be monitoring on crossfire. 300% is maximum lq, 100% is when it switches to 50hz, and you do not need to turn around till 70%.
LQ is always higher than RSSI. RSSI is a logarithmic function, while LQ is a strong exponential. So, LQ remains at a high percentage and then drops off rapidly. RSSI drops off rapidly first, and then decreases more and more slowly. at 50% of your range, LQ will be 100% and RSSI will be at 10-15%.
Previously on Betaflight, the LQ would only show a maximum of 99%, you still turned around at 70% but you couldn't see the fluctuations between 150hz mode and 50hz mode reliably. Now that we have 300% and the new 0:x 1:X  2:X to display, you can see the fluctuations. additionally, keep in mind that small losses in 150Hz mode are perfectly normal. they should remain within 10% on normal flying.
LQ on Betaflight
This is confusing a lot of people new to crossfire and with the recent changes to betaflight  even some experienced pilots seem to be confused by what the Betaflight OSD is telling them.
There are 2 different ways of showing LQ on Betaflight
#1 the original way (BF 3.5.7 or earlier and Current DJI)  it uses the Betaflight osd element “Rssi” and you would send LQ down a channel from the rx. This is actually the Current way to do it on a DJI video system, this will show a maximum LQ of 99% turn around is 70%
https://youtu.be/3542fuE_bag
#2 the new way post Betaflight 4.0 - you do not need to change any setting on the channels, just hook up crossfire how you normally would to the FC and betaflight can read your LQ automatically
Betaflight has had 3 different ways of showing LQ
#1 3.5.7- DJI - this will show as a metric to 99% turn around is 70%
#2 BF 4.1 - this will show up to 300% turn around is 170%
#3 BF 4.2 and later - this is a tad more complicated but gives great information
0: 70-100 (4hz mode, Forced Telemetry enabled. DON’T DO THAT)
1: 100 (50Hz mode)
2: 100 (150 Hz mode)
What this shows is the RF mode you are in and the % of that mode. For example 2:100 would be full signal at 150hz (when you see a 2:x you don't need to worry about your signal at all, you are in 150hz mode and have TONS of range left)  . 1:100 would show you switched to 50hz (this will happen moderately early at range, do not be worried, the main range beyond around 2-5 miles is in 50hz. Turn around for this would be 1:70 on the Betaflight OSD. This means you are at 70% LQ and it is time to turn around.

Pairan, 2020-07-15:
> Also,i beleive that you just need to use the LQ osd item and it will show correctly.

As far as I can see there is no such thing within iNav 2.5.1 - Yet I will RSSI_SOURCE = PROTOCOL a shot