iNavFlight/inav #4563: GPS lock "lost" with 12 Sats

MartinHugh, 2019-03-29:
I recently maidened a 1m fixed wing.

The GPS is in a standard position away from other electronics, and the same configuration with the same wing type but with a different FC has worked fine a few years.

I have "Min. GPS sats for valid fix" = 6

After powering up the system acquired 12 satellites.  Despite this it was reluctant to arm (navigation not safe).  I waited for a while and it eventually decided it was safe (still 12 Sats showing on OSD).

When flying occasionally the Sats display would flash and the Sat related metrics (e.g. distance to home point) would stop updating.  After a while the flashing would stop and the metrics would refresh.

What do I need to do to be able to fly with 12 satellites ?

teckel12, 2019-03-29:
@MartinHugh The number of satellites does not at all relate to the lock. A GPS lock has to do with dilution of precision (DOP). This is shown in INAV as HDOP, but in most cases its really PDOP (which is a 3D dilution of precision).

Without getting too technical, if the satellites are all in one direction and not spread across the sky, you could have 12 satellites but not enough precision for a 3D lock.

Probably a good idea to show the HDOP on your OSD to get more relevant information. The number of satellites really isn't a good measurement of a 3D fix at all. HDOP/PDOP, however, is a much better value to monitor. I wouldn't fly unless the HDOP/PDOP is under maybe 2.5. If you can't get that, there's probably some kind of hardware/interference issue (like maybe a RunCam Split).

MartinHugh, 2019-03-30:
@teckel12 thanks for your comments, will look into that.

MartinHugh, 2019-03-31:
@teckel12 thanks for your wise words. I added "HDOP" onto the OSD which is very useful, as it showed a very high HDOP value regardless of the number of sats collected.
I then swapped the BN-220 antenna module for a replacement and very quickly got 9 sats@1.3 "HDOP".
It seems that for now this one is sorted.
Many thanks.

closed 2019-03-31 by giacomo892