iNavFlight/inav #4564: Max bank angle in Angle mode

MartinHugh, 2019-03-30:
I notice a newly set up flying wing that the maximum bank angle is very slight, like around 30-40 degrees.   What setting should I use to increase this (to say 60 or 80 degrees)

Thanks

shellixyz, 2019-03-30:
https://github.com/iNavFlight/inav/wiki/Fixed-Wing-Guide#step-4-replace-default-values

closed 2019-03-30 by shellixyz

MartinHugh, 2019-03-30:
@shellixyz thanks. Cant believe I missed these angle_inclination settings. I did all of the others but for some reason these two appear to be still at defaults.

MartinHugh, 2019-03-30:
Actually - I confirmed this by looking at the last diff I had.

But on checking the FC itself - these were set to 600 in both cases.

Also checked that all the other settings referenced had been set to sensible values

What can I check next?

Do I need to recreate another issue now this one is closed ?