iNavFlight/inav #4565: Max bank angle in Angle mode - angle_inclination set

MartinHugh, 2019-03-30:
Please refer to https://github.com/iNavFlight/inav/issues/4564 which was closed before I had a chance to check the suggestion made (Fixed Wing guide, Step 4).

The values had in fact been set to sensible values,

# get max_angle_inclination_rll 
max_angle_inclination_rll = 600
Allowed range: 100 - 900

# get max_angle_inclination_pit 
max_angle_inclination_pit = 600
Allowed range: 100 - 900

What other reasons would I have restricted angle particularly noticed in the roll axis ?

Thanks

shellixyz, 2019-03-30:
Please post a CLI diff. Do you have a log ?

MartinHugh, 2019-03-31:
So sorry I have no log (and FC does not have embedded flash for log or SD card).

Diff follows, thanks for looking 

```
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/MATEKF411 2.1.0 Feb 25 2019 / 17:02:49 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 3 1000 2000 1500 -100

# feature
feature MOTOR_STOP
feature GPS
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 1 2 115200 38400 0 115200

# led
led 0 2,15::AT:1
led 1 3,15::G:1
led 2 4,15::G:1
led 3 5,15::AT:1
led 4 10,15::CT:1
led 5 11,15::CT:1
led 6 12,15::CT:1
led 7 13,15::CT:1
led 8 6,6::CT:1
led 9 7,6::CT:1
led 10 8,6::CT:1
led 11 9,6::CT:1

# color

# mode_color
mode_color 6 0 0
mode_color 6 1 2

# aux
aux 0 0 0 1300 2100
aux 1 1 1 900 1275
aux 2 1 2 900 1300
aux 3 3 1 1300 1700
aux 4 3 2 900 1300
aux 5 9 1 1300 1700
aux 6 9 2 900 1300
aux 7 8 2 1725 2100
aux 8 10 2 1300 1700
aux 9 14 6 1300 2100
aux 10 27 4 900 1225
aux 11 28 4 1800 2100
aux 12 11 3 1300 2100

# adjrange

# rxrange

# temp_sensor

# osd_layout
osd_layout 0 0 5 2 V
osd_layout 0 1 23 2 V
osd_layout 0 2 0 0 V
osd_layout 0 7 13 13 V
osd_layout 0 9 1 2 H
osd_layout 0 11 24 1 V
osd_layout 0 12 13 1 V
osd_layout 0 13 1 7 V
osd_layout 0 14 4 1 V
osd_layout 0 15 24 7 V
osd_layout 0 20 17 13 V
osd_layout 0 21 1 13 V
osd_layout 0 22 14 11 V
osd_layout 0 23 15 14 V
osd_layout 0 28 1 3 V
osd_layout 0 30 1 12 V
osd_layout 0 31 8 1 V
osd_layout 0 34 10 10 V
osd_layout 0 38 26 3 H
osd_layout 0 40 21 14 V
osd_layout 0 41 24 10 V
osd_layout 0 42 24 9 V
osd_layout 0 45 0 0 V
osd_layout 0 86 10 14 V
osd_layout 0 87 0 11 H

# master
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 40
set acc_hardware = MPU6000
set acczero_x = 103
set acczero_y = -43
set acczero_z = -276
set accgain_x = 4099
set accgain_y = 4097
set accgain_z = 4028
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 10
set min_throttle = 1149
set max_throttle = 1900
set failsafe_delay = 0
set failsafe_off_delay = 0
set failsafe_procedure = RTH
set align_board_roll = -8
set align_board_pitch = 70
set platform_type = AIRPLANE
set model_preview_type = 8
set 3d_deadband_low = 1425
set 3d_neutral = 1475
set small_angle = 180
set gps_provider = NMEA
set gps_sbas_mode = AUTO
set nav_rth_allow_landing = NEVER
set nav_rth_alt_mode = FIXED
set nav_rth_altitude = 7620
set nav_fw_max_thr = 1750
set nav_fw_bank_angle = 55
set ledstrip_visual_beeper = ON
set osd_units = UK
set osd_crosshairs_style = AIRCRAFT
set mode_range_logic_operator = AND

# profile
profile 1

set fw_p_pitch = 6
set fw_i_pitch = 10
set fw_ff_pitch = 52
set fw_p_roll = 6
set fw_i_roll = 8
set fw_ff_roll = 49
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set tpa_rate = 20
set tpa_breakpoint = 1600
set roll_rate = 28
set pitch_rate = 14

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 320
set vbat_warning_cell_voltage = 330
set battery_capacity = 4000
set battery_capacity_warning = 1400
set battery_capacity_critical = 1000
```

shellixyz, 2019-03-31:
Can't see anything that would restrict the bank angle. Could you share a recording of your FPV feed if you got FPV on this wing ?

giacomo892, 2019-03-31:
Be sure to have ROLL angle indication on the OSD enabled

MartinHugh, 2019-03-31:
@shellixyz thanks for checking.  Will try to do a recording next time I am out.
@giacomo892 thanks for the hint - will check this.

AnandNL, 2019-04-03:
could it be this one: `nav_fw_bank_angle = 20`  since its 20 by default and so will not show in the diff?

shellixyz, 2019-04-03:
`nav_fw_bank_angle` have no effect on angle mode, only on navigation modes RTH, CRUISE and WP

MartinHugh, 2019-04-06:
A bit of empirical evidence goes a long way.
With roll angle displayed, today's flight showed that full bank was holding 58 degrees (with angle inclination set)

Fault traced to somewhere in between goggles and transmitter :-)

Thanks to all.

shellixyz, 2019-04-06:
Very good :wink:. Thanks for the feedback

closed 2019-04-06 by shellixyz