iNavFlight/inav #4568: Detect failsafe and aux channels

hali9, 2019-04-01:
## Current Behavior
Failsafe is detected only by 4 first (flight) channel.

## Desired Behavior
Add the optional possibility to detect an incorrect pulse on the aux channels, and optional set up predefined value.

## Suggested Solution
Create `rxaux` (CLI) array settings for aux channels:
`rxaux  0 1400` - channel 5 (aux 1) can cause failsafe and if failsafe is triggered, aux1 have value 1400
`rxaux  1 1` - channel 6 (aux 2) can cause failsafe and if failsafe is triggered then aux2 value is hold
`rxaux 2 0`  - channel 7 (aux3) default value 0, feature disabled, failsafe ignore aux3 and it value is hold (works as now).

## Who does this impact? Who is this for?
* Anyone who can not set values ​​for failsafe in their transmitter for all or some channels.
* In the case of using receivers where, after losing connections, all or some channel values ​​are random.
* Turning on or off the specified box after going into failsafe (for example disable flaps, camstab, enable blackbox etc).
* Prevent unexpected behavior. When flight channels are valid and one or some aux channels are invalid. And this invalid aux was changed fly mode.
 * Setting of aux channels in the desired values when  failsafe is start manual (BOXFAILSAFE enabled).

## Additional context
Additionally: aux can not trigger failsafe, but in the case of failsafe it have defined value: 
`rxaux  3 -1400` 

Regardless of the above, it would be good to see the inav-configurator in addition to the raw value (as it is now), the date value. So maybe for profile 1 raw value, profile 2 raw/data switching by 1 seconds, profile 3 data value.

A similar solution was worked once when failsafe was divided into stage 1 and stage 2. This has been removed in https://github.com/iNavFlight/inav/pull/1596

Also RSSI is now using raw values, so good point is when rssi channel is set up in rxaux it overwrite rssi value.

What do You think about this, comments are welcome.

lichtl, 2019-04-01:
Why not use the existing "failsave" mode. You can apply this to any aux channel already

hali9, 2019-04-01:
> Why not use the existing "failsave" mode. You can apply this to any aux channel already

I use and I want use existing failsafe, but current failsafe ignore aux and only hold it using last good value .

digitalentity, 2019-04-01:
Stage 1/2 failsafe and ability to configure failsafe channels at will was removed for a simple reason - to simplify failsafe setup. This is a critical system and users were making (and sill make) too many errors. 

I'm very reluctant to be adding more complexity to failsafe.

Jetrell, 2019-04-03:
> 
> 
> Stage 1/2 failsafe and ability to configure failsafe channels at will was removed for a simple reason - to simplify failsafe setup. This is a critical system and **users were making (and sill make) too many errors**.

I think stage 1/2 may have a place in non GPS guided flight control systems. While in iNav I don't see the need anymore for them.  
The main reason that users are still making failsafe setup mistakes, is because the Wiki isn't totally clear about things; like the way stage 2 interacts with stage 3, using stick Arming.  Which it should not do under any circumstances.

closed 2019-06-20 by stale[bot]