iNavFlight/inav #4569: iNav FC as OSD only - no home direction / home distance possible

toontoon1, 2019-04-01:
Sorry, If I'm posting in an appropriate way - but this is my first github post..
I'm using a Matek F411 wing with Inav 2.1.0 as an OSD only (only GPS, CAM and VTX connected, NO servos, ESC, RX). On the OSD I want to see the direction (arrow) and distance to home - which is not shown at the moment (arrow doesn't react, no distance to home displayed). As far as I understand (with the help of rcgroups) , the FC must be armed for that - which is not possible, because no Rx is connected. I tried to give modes/arm the whole span (900-2100) and 'set  nav_extra_arming_safety=OFF', but without success.
Is it possible to get a feature when using an FC as OSD only, to arm the FC without having an RX connected or to set a home point just when powering the FC (without arming)?

giacomo892, 2019-04-01:
I'd give a look at MW OSD. Cheaper and effective.

toontoon1, 2019-04-01:
Tx - but at the moment I would like to stay with an iNav FC and use iNav's OSD features...

giacomo892, 2019-04-01:
Then what about removing your current FC without OSD and replace it with the F411 wing you have?

toontoon1, 2019-04-01:
The F411 wing is mounted in a ZOHD Orbit which has an origjnal stabilizer (=FC) included (3 flight modes, no RTH). This stab works very very well, so for the moment I'm trying to keep it. And I have only a 4 channel RX installed in the Orbit (Spektrum Ar410) and all its channels are occupied by the ZOHD stab. So no free channel available to arm the F411.
Maybe some time later I will start to use the F411 for what it is made for - a full featured FC. But at the moment it should be just an OSD... :-)))

toontoon1, 2019-04-01:
Sorry, just pressed the wrong button (close and comment) and had to reopen...

toontoon1, 2019-04-02:
Forget what I said regarding 'free channel' - stupid talk... I can't connect an Spektrum Ar (like Ar410/Ar620) to the F411, as only SBUS or satellite is possible. So no realistic chance to get the FC armed via the RX - I need a new feature 'set/show home point data without arming'

rrk1, 2019-04-07:
toontoon1, are you sure you can not connect F411's RX pin in parallel to you existing controller RX? F411 seems to support at least PPM protocol.

toontoon1, 2019-04-07:
How would I do that? To what port at the F411 and from what port at the Ar410 (all ports are occupied and from which port should I use the signal as parallel? One port throttle, one aileron, one elvator, one mode switch. Can you show me a drawing please?
At the moment as a workaround I connected a spektrum Satellite to the F411 and had it bind to a new modelfile at my DX9. then at the field use this modelfile with an switch to arm th F411 (home point set), switch to the Orbit's modelfile at the DX9  (F411 modelfile goes into failsafe, iNav failsafe it set to 'do nothing') and everythign seems to be ok - home point and distance is ok during flight. But that is uncomfortable....  A feature 'set home point without arming just with power on' would be much better....

rrk1, 2019-04-07:
Well, you have 3 wires on each channel of RX. Signal, ground and +5V. You splice signal wire and run it to F411 input too. I'd try first to use throttle channel to use as INAV ARM. However, it is possible that you could not route route some AUX PWM RX channel on a spare pin of F411 board. Do not have this board, no way to check, sorry.

What will definitely work is to try to find an old PWM -> PPM converter board. They were popular some time ago with Arducopter or Pixhawk controllers to adapt them to old school PWM RXs like yours. You hook converter inputs to your RX to be able to simulate stick arm command for INAV, and connect converter output to PPM input of Matek. That should definitely work.

toontoon1, 2019-04-07:
TX for your help!
Sounds complicated... I also do not know whether the F411 wing allows PWM input, will have to check that. But with the hassle of adding a pwm-ppm converter and splitting the RX signal to this adapter, I (IMHO) can also stay with my workaround with the satellite and a second modelfile - as long as there is no feature update at iNav :-)))))
I've also ordered a MWOSD with an external baro - maybe that will be the easiest way to go, when wanting to switch the OSD between different planes. But nevertheless I would still like to prefer to use my different iNav boards for such a standalone OSD. I have several of them and that would be a good training step to go into further iNav settings which would probably end using iNav as a complete FC. I will make this decision dependable on the way how my feature request will be handled - if there will be no option to get an iNav FC showing home point and distance without the need of arming, maybe the MWOSD is my future favourite...

toontoon1, 2019-04-12:
BTW: How can I change the label of my issue? 'giacomo892' has set it to 'support', but I want to have it set ot 'feature request' - how can I do that? The github 'help' tells my something about 'selecting a checkbox', but I do not see any checkbox at my issue.....

DzikuVx, 2019-04-24:
INAV is a flight controller firmware, not OSD overlay system and such we will rather not implement a feature that alllows to display home arrow while not controlling the airplane. If you want such a thing, you have to use a flight controller as a flight controller with the information that it was armed and where. Thanks

closed 2019-04-24 by DzikuVx