iNavFlight/inav #4571: Feature request: analog temperature sensor support

Jetrell, 2019-04-02:
I would like to request this option. To keep the weight and size down in smaller models. That also use smaller devices like ESC's.
In the case of an LM75 breakout board. It is almost as large as the ESC, which blocks its cooling efficiency.
While a thermistor is very small and can be slipped under the heat shrink tube of the ESC, with ease.

The RSSI ADC pin on most Flight Controllers that is not used by many of us (who put RSSI/LQ on a radio channel) could be an option for this.
And it only uses 2 wires instead of four. To simplify the application.

[image]

teckel12, 2019-04-02:
@Jetrell There are very small 1-wire digital temp sensors that don't need any kind of breakout board. I've used analog LM75 and they're crappy compared to the 1-wire digital ones. I believe INAV supports 1-wire temp sensors.

Jetrell, 2019-04-02:
@teckel12 The TO-92 cased DS18B20 is at top in the picture, above the 10K NTC thermistor. 
It is also small. But at present it requires a DS2482 I2C interface connection for it/them to function.

@shellixyz said he is also thinking of adding a 1-wire UART interface for the DS18B20. I assume it will then be stand alone.

My idea for the thermistor was based around small lite weight FPV models that have minimum size and weight. Even sub 250grams plane/quads. While still being able to use the temperature function.

teckel12, 2019-04-02:
@Jetrell But analog temp sensors are crap IMHO. I'd wait for 1-wire. I use that for other projects and it works great and small.

Jetrell, 2019-04-02:
I do agree Tim.  Thermistors are not ideal. They do have drift, the farther the measured temperature gets above or below their factory calibrated 25°c.
But they still give a readable results. Under my tests....A drift of 5% at 85°C ESC temperature is still letting us know that the ESC is running hot and should be watched.

Unless you know anyone that makes a digital temp chip in a TO-92(S) case or a SOT-23 case. Then a thermistor is the smallest temperature sensing device that I know of.

wx4cb, 2019-04-02:
here's a couple of small to92 cased sensors... I would think the lm19 would be closest to what you want

http://www.ti.com/product/LM19

the microchip MCP9700 has a humidity sensor too and can also be gotten in the to92 case form factor from allied
https://www.microchip.com/wwwproducts/en/MCP9700

tmp359: 
https://www.digikey.com/product-detail/en/analog-devices-inc/TMP35GT9Z/TMP35GT9Z-ND/996945&

rrk1, 2019-04-05:
I am strongly in favor of "befriending" DS18B20 and likes with INAV. Someone just needs to adapt 1-Wire protocol to run directly on a spare pin of flight controller. The protocol itself is very simple and timing-relaxed.

Standard NTC are very convenient and cheap (I recommend to look for ones from refrigeration thermostats, looks like small epoxy drop on the end of PVC cable - small and waterproof). However NTC needs a precision loading resistor and shall be individually calibrated. Also, transfer curve is non-linear and differs between types. And a random loading resistor from junk box may not work too because carbon film resistors have huge temperature drift themselves (was bitten by this recently when tried to do a simple battery voltage sense divider). Maxim 18B20's are instead already calibrated from the box to about +- 1C, saving MAJOR hassle, and packaged in to-92 case.

rrk1, 2019-04-05:
Also, what we can do is to add an universal ADC data input  "class".

That way we can assign an ADC input to a few easily understandable configurable conversion steps and made it useful for almost all analog sensors possible - temp, voltage, currrent etc... 

Roughly like:

step 0: get ADC data
step 1: add +- offset1 (mostly for ADC drifts, leaks, etc)
step 2: solve R1/(R1+R2) ratio (optional, for resistive sensors,  thermistors and like)
step 3: apply +- offset 2
step 4: multiply to +- slope(scale)
step 5 round to N digits
step 6 add ASCII symbol(s) (A, C, V etc)
step 7 display on OSD, optionally uplink as telemetry.

rrk1, 2019-04-05:
And some kind of lowpass filtering (at least fixed to about 0.5Hz) shall be here too.

teckel12, 2019-04-06:
@rrk1 This is why I don't like analog temp sensors. I think it would be a better use of time to add a 1 wire protocol rather than analog nonsense.

Jetrell, 2019-05-01:
@shellixyz  due to there being much dislike for analog sensors, even though they are the smallest and lightest type available.  
Would there be any chance of a 1-wire protocol for the DS18B20, using soft serial?
_Recapping_ .... My hope was for an addition to the temperature function, for a small sensor to be implemented with simplistic wiring, for use in small light models. The type of FC used in such models will also have limited IO's.
Due to the world laws getting tighter and tighter on FPV.  Sub 250g is becoming more common. So keeping the weight to a minimum is also very important.

You did an awesome job of the temp sensor function.  I use it in my larger planes and quads, But I would also like to use this function in my smaller planes as well.  

The only other possibility may be. If Dshot telemetry is implemented in iNav, for direct ESC temperature sensing...and RPM too. But its limited in usable hardware!

shellixyz, 2019-05-02:
> Would there be any chance of a 1-wire protocol for the DS18B20, using soft serial?

Yes softserial or UART. It is in my todo list.

Jetrell, 2019-08-04:
> 
> 

> 
> Yes softserial or UART. It is in my todo list.

Not rushing you Shelli. I just thought I'd mention it again in case it slipped your mind :-)

closed 2022-07-05 by Jetrell

dcan999, 2024-12-22:
I see this was closed as completed, but it seems it was never implemented? The instructions for using a DS18B20 still say the DS2482 is required.