iNavFlight/inav #4573: cli map

MrScott66, 2019-04-03:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)

MrScott66, 2019-04-03:
Sorry for dumping out my question's text in the subject line, this web site's layout makes little sense to me

DzikuVx, 2019-04-03:
I do not understand. What does INAV output has to do with TEAR or X4R? You want to connect flaps and retracts to what? Flight controller? X4R?

MrScott66, 2019-04-03:
Is there a way to have iNav look for throttle/elevator/aileron/rudder (TEAR) on channels other than 1 through 4?

I want to connect flaps and retracts to my receiver but the TEAR output was on channels 1 through 4.  So on my Taranis I moved TEAR to output on channels 5 through 8 and assigned flaps and retracts to outputs 1 and 2.  However, iNav only looks for TEAR on outputs 1 through 4. 

Is there a way to change TEAR to 1234TEAR like these guys?

https://www.rcgroups.com/forums/showthread.php?2567258-x4r-sbus-using-3-unused-pins-of-3-16-channels/page2

Thank you,
Scott

DzikuVx, 2019-04-03:
You can bind X4R in 9-16 mode and then SBUS pin will still give 1-16 but X4R pins will output channels 9, 10 and 11

closed 2019-04-03 by DzikuVx