iNavFlight/inav #4574: FC F35 compass erorr!!

letruong6006, 2019-04-04:
hello, can someone help me? I am using the FC furiousvF35 for X-UAV talon. I noticed the circuit of the circuit was only in the wrong direction. I checked it with the compass direction of the phone with the direction of the FC circuit. (confirm the direction of the phone is correct). But the direction of the circuit is different, I'm sure the direction of FC is only wrong, I tried calibrate compass many times but the result is still the same. Board and Sensor Alignment parameters are default. what should i do? :(

teckel12, 2019-04-04:
@letruong6006 Is your flight controller pointed forward? Is the compass on the GPS module or on the FC?

wx4cb, 2019-04-04:
the flight controller is 180 degrees out if the pins are pointing to the front of the aircraft (hard to tell which direction it's installed. Also, you don't need the compass on a fixed wing so you can disable it.

letruong6006, 2019-04-05:
> @ letruong6006 Bộ điều khiển chuyến bay của bạn có hướng về phía trước không? Là la bàn trên mô-đun GPS hoặc trên FC?

The controller flies forward. The compass is definitely on FC. GPS does not have a compass. I had some great flights with stable parameters. but something made it go wrong. while RTH flies around on his head. It has fallen. and then the compass is no longer accurate

letruong6006, 2019-04-05:
> bộ điều khiển bay lệch 180 độ nếu các chân chỉ về phía trước máy bay (khó có thể biết được nó được lắp theo hướng nào). Ngoài ra, bạn không cần la bàn trên một cánh cố định để bạn có thể vô hiệu hóa nó.

How to turn off the compass with CLI? My plane crashed after an accident. I took the plane in my hand and walked about 50m. The arrow pointing to the house is misleading, even if I move it to the home position, it points in a different direction, all parameters are not changed as before the accident. What can I do?

wx4cb, 2019-04-05:
Turn it off in the configuration tab. Set magnetometer to none

teckel12, 2019-04-05:
@letruong6006 In the configurator, just above where you circled the rotation adjustment, you can set compass to none.

letruong6006, 2019-04-05:
Red circle arrow below. does it work when i turn off the compass? I really need it to be able to control in the foggy days
[image]

wx4cb, 2019-04-05:
Yes it will work fine as it will come from the gps

teckel12, 2019-04-05:
@letruong6006 Since there's always forward movement with fixed wing, heading is calculated by using GPS coordinates.

Multi-rotor needs a compass because it can be totally stationary, so direction can't be calculated with just GPS.

So as long as your fixed wing is moving forward, it will have accurate heading.

letruong6006, 2019-04-05:
I will try it today. Thank you Teckel12 and wx4cb very much :)

closed 2019-05-09 by DzikuVx