iNavFlight/inav #4575: FrSky R9 Slim+ with Flex Firmware no Telemetry

philbertphotos, 2019-04-04:
## Current Behavior
No telemetry on any data not even RSSI using Flex with or without Fport

## Steps to Reproduce
1.Installed old firmware and cant get Telemetry 
2. Installed Flex with and without Fport and nothing works

## Expected behavior
Telemetry Data is not transmitted to the controller tired other suggesting with similar issues that have been closed to no solutions.

## Additional context
With all the changes I always got RC signals to work as they should. 

- FC Board name and vendor: Matek f722 Standard 
- INAV version string: 2.10

teckel12, 2019-04-04:
@philbertphotos What port on the FC are you connecting to? Are you using the inverted or uninverted connection on the Rx? Include a CLI dump so the inversion/duplexing options are known.

sincoder, 2019-04-07:
with FPORT works fine