iNavFlight/inav #4576: new OSD functions

Ralfde, 2019-04-04:
The inav software is very 

When the airplan main functions like ports RX signal are set, sometimes additional settings need to be done on the flying field without having a PC available.
It's sometimes needed to add and delete mixers and to adjust the weight.
Also it's needed to change the servo trim or servo travel.
It would be very nice, if these setting would be also possible in a OSD submenu, so there is no PC needed on the flight field. Also sometimes the USB plug is not easy to reach.
[image]
[image]

It would be a really nice feature for one of the future inav updates.

I wish also the CMS-HoTT mode will be included to the future versions.

fiam, 2019-04-28:
@Ralfde Check https://github.com/iNavFlight/inav/pull/4616 for CMS over HoTT

Scavanger, 2019-05-20:
See #4724

closed 2023-10-27 by DzikuVx