iNavFlight/inav #4583: Remove support for stick arming

digitalentity, 2019-04-06:
Stick arming comes from the ancient era of RC systems where we didn't have enough switches and sometimes we had no switches at all.

This is no longer the case - even the cheapest RC systems have at least 6 channels (more often more than 8) where you can dedicate a channel to arming.

Stick arming brings along a bunch of hacks to the code to accommodate for different behavior at zero throttle. Sometimes it causes unexpected behavior - i.e. trying to do a full  yaw at zero throttle. And if combined with MOTOR_STOP this is becomes even less safe.

DzikuVx, 2019-04-06:
Yes, I agree. It's time to let legacy things like this go

teckel12, 2019-04-06:
Even my first $50 Tx/Rx had 8 PWM channels, enough to arm with a switch. I say remove it, I find it only dangerous.

rrk1, 2019-04-07:
Have a nice scar on my hand back from the times of early BF and stick arming ;-)

leeph, 2019-04-09:
I started my autopilot journey with EagleTree Vector - it's never been an option with that system and I was kind of surprised to see it in iNav at all. Goodbye stick arming! We will not miss you.

ChristianFFB, 2019-04-10:
I will not miss it. Dangerous on planes.

DiscoMan18, 2019-04-16:
I vote for leaving it intact, at least for multicopters. Switchable on/off via CLI, stick_arming = OFF by default.

flybabo, 2019-04-17:
I also have a scar on one of my left fingers.
I vote for removing - we can't carry all the baggages forever .

KiwiDavid, 2019-04-24:
I agree with DiscoMan18.   There must be many beginners and students on limited budgets using something like the FS-i6  (6 channel tx).  If switch arming is mandatory it leaves them a single channel for flight modes.  While switch arming would be nice I can't see how stick arming is dangerous unless you're fumbling with the craft in one hand and the tx in the other.  Perhaps I've just been lucky all this time not to have shared the same air space as a rotating prop.

DzikuVx, 2019-04-24:
I do not find argument of FS-i6 a really valid one. Even without channel mod 1 switch for arming and second for flight modes is enough. If you buy the cheapest hardware you can not really expect full support for it. 
Besides, FS-i6 is $41 while 10CH FS-i6X is $42.  So no, it's not really a very important argument.

giacomo892, 2019-04-24:
Also you can easily mod the FS-i6 to run 10ch on IBUS. @KiwiDavid

digitalentity, 2019-04-24:
If you're really on a budget you can go below $40 with [Radiolink T8FB](http://s.click.aliexpress.com/e/JXqQOWc). Using a 6ch transmitter is really not a valid justification to keep a dangerous feature. This PR also is not making previous INAV versions unavailable.

rrk1, 2019-04-24:
Well, IMHO it is not stick arming itself is inherently dangerous, but current very non-selective implementation in BF/INAF. It is a poorly designed relic back from Multiwii times and was never given a critical thought in all those years. DJI NAZA does it in a proper way for example, with two sticks in the corners, and I never had myself nor heard about accidental arming / disarming problems with it.

Anyway, IMHO again stick arming has become incompatible with modern flying style where a lot of maneuvers are now done at zero or near zero throttle, so it have to die anyway... :)

KiwiDavid, 2019-04-24:
OK they're all good points although the price differential between 6ch and 8ch hardware is of little comfort to an existing user base.  Perhaps the new switch arming release could include a note indicating that a minimum of 5 ch are required for basic operation before the allocation of any additional flight modes or features.

closed 2019-04-24 by DzikuVx

padcom, 2019-06-20:
+1 for finally getting rid of it. I did have a 5+ km flight once that went down because of stick disarming and I never want to have that surprise again.

padcom, 2019-06-20:
As for the price difference... C'mon! If you want to fly a serious aircraft it is going to cost you a little extra. Invest in X9D or at the very least Turnigy 9x and you'll be all set to go. Remember the old rule: who buys cheap pays twice.

DiscoMan18, 2019-06-20:
> +1 for finally getting rid of it. I did have a 5+ km flight once that went down because of stick disarming and I never want to have that surprise again.

Isn't it easier to disarm your aircraft (or arm it instean) by accidentally flicking the switch? As to me, it's even more dangerous...

teckel12, 2019-06-21:
@DiscoMan18 You can prevent that as well, you can set the disarm switch to only work when throttle is at zero. I'm on my phone and the CLI command escapes me at the moment.

PR #4640 (2019-04-24): Remove sticks arming
closes #4583