iNavFlight/inav #4587: Add Props-in(reversed prop direction) to mixer for quads

jumpingcloud, 2019-04-07:
## Current Behavior

There is no Mixer Preset or Mixer option to let iNav know there is a Props-In ( reversed prop direction ) propeller orientation on a Quad Copter.  

## Desired Behavior

Need a way to let iNav know a user is running Props-in ( reversed prop direction )

## Suggested Solution

Add to the mixer panel a props-in ( reversed prop direction ) setting for quads.

Alternatively:
Add, in the documentation, a mixer setting for a props-in setup. 

## Who does this impact? Who is this for?

Quadcopters running props-in. Some frames ( fold-able frames ) require this.  And some frames fly better with this orientation. ( Ducted  props )

## Additional context

Prop props Propeller direction reversed

If there's a way to do this from the mixer I couldn't find it anywhere.

giacomo892, 2019-04-07:
Issues regarding presets / mixer presets have to be opened in the inav-configurator repo.

You can just adjust the QUADX mixer by swapping all the signs in the yaw.

closed 2019-04-07 by giacomo892

digitalentity, 2019-04-07:
Also, `yaw_motor_direction` changes the expected direction of motors. Just `set yaw_motor_direction = -1` with default QUADX mixer and you are good to go.

giacomo892, 2019-04-07:
@digitalentity wondering if we should drop that setting now that we have a so flexible mixer