iNavFlight/inav #4588: FFPV VTX control OSD menu appearance

vicdzen, 2019-04-07:
## Current Behavior
[image]

Wrong Page name and invisible band in **Band** line
## Steps to Reproduce
Just move to Features -> FFPV VTX in OSD menu

## Expected behavior
Correct and visible information in OSD menu page 

## Suggested solution(s)
Page name should be something like **FFPV VTX**
Band line should indicate band name ("2.4G A" or "2.4G B")

## Additional context

---

- FC Board name and vendor: Matek F411 wing
- INAV version string: 2.1.0
- VTX FW version: VTX2G4_FCC 05 Apr 2019

vicdzen, 2019-04-07:
Also [This one](https://github.com/iNavFlight/inav/issues/4510) is related to FFPV protocol

closed 2019-06-20 by stale[bot]

KeenFPVer, 2019-07-08:
I have same issue using same FC and FFPV VTX2.4
Mine also does not change power every time I can see this by performance and he RED LED on the VTX does not blink 3X for 500mW
[image]