iNavFlight/inav #4590: [F35 FC] OSD does not display on the screen

letruong6006, 2019-04-08:
I used the Fc F35 circuit, some of the previous flights were great but suddenly OSD was not displayed on the screen, no parameters were changed before, OSD in INAV is still activated, the latest version 2.1 .0. I need help
[image]

parchedpacha, 2019-04-08:
I recently had this issue, and I solved it by uploading a new font file. Maybe that will work for you too? if it doesn't work immediately try rebooting after the upload is completed.

shellixyz, 2019-04-22:
If it was working before and nothing changed in the software then it seems to me that it can only be a hardware issue